mavlink/qgroundcontrol #8486: Minimum window size is too large.

RicardoM17, 2020-03-05:
## Expected Behavior
The QGC window can be reduced up to very minimal sizes (even if not all elements of the GUI are visible)

## Current Behavior
The window can only be reduced up to a certain amount which is much larget than it should be as there is still plenty of space.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open latest QGC.
2. Try to reduce window size by dragging a corner.
3. It only works up to a certain point.

## Log Files and Screenshots
[image]

From the screenshot above you can see the minimal size that I can currently set the QGC window to. There is plenty of space it takes way more than half the screen in both directions. Unless I'm missing a setting this is a bug that should be fixed. Not sure if other users are experiencing this as well.

DonLakeFlyer, 2020-03-11:
> From the screenshot above you can see the minimal size that I can currently set the QGC window to.

@RicardoM17 Sorry but from the screenshot I can't see any of that. In order to see that I need to have a screen shot of the entire screen. And then also can you tell me resolution of your screen.

RicardoM17, 2020-03-11:
Hi @DonLakeFlyer You are correct in that. My point was to show that in terms of GUI there is still plenty of space left and as such no reason toe prevent the window from becoming smaller.

That being said I attach two full screen shots that show the minimum size for the new version, compared to what I usually use which is 1/4 screen. However it should be noted that in this version the screen size has pratically no lower limit, so it can even be resized to something much much smaller.

New Version:

[image]

Old Version:

[image]

II discovered this issue when updating to the new version but I believe this was introduced much earlier. Problem is replicated on my other machine.

Both screens are 1920 x 1080p standard resolution.

DonLakeFlyer, 2020-03-11:
Yeah, that seems much to big.

@dogmaphobic As far as I can tell this code is new in 4.0:
```
ApplicationWindow {
    id:             mainWindow
    minimumWidth:   ScreenTools.isMobile ? Screen.width  : Math.min(215 * Screen.pixelDensity, Screen.width)
    minimumHeight:  ScreenTools.isMobile ? Screen.height : Math.min(120 * Screen.pixelDensity, Screen.height)
    visible:        true

    Component.onCompleted: {
        //-- Full screen on mobile or tiny screens
        if(ScreenTools.isMobile || Screen.height / ScreenTools.realPixelDensity < 120) {
            mainWindow.showFullScreen()
        } else {
            width   = ScreenTools.isMobile ? Screen.width  : Math.min(250 * Screen.pixelDensity, Screen.width)
            height  = ScreenTools.isMobile ? Screen.height : Math.min(150 * Screen.pixelDensity, Screen.height)
        }
    }
```

I don't think 3.5 code set a minimum. I would be whatever Qt decided the mins were for the old QMainWindow. Which ended up allowed to go fairly small.

I'm not quite sure what this code is targetting for a minimum but however it works for this csae that's too big for a minimum size.

PR #8590 (2020-03-24): Adjust minimum application window size
In response to https://github.com/mavlink/qgroundcontrol/issues/8486

I'm using an arbitrary minimum size where the UI _mostly_ fits. Needs testing on other platforms. Especially when connected to low DPI monitors.

[image]

Closes #8486
closed 2020-03-24 by dogmaphobic