mavlink/qgroundcontrol #8490: Can't connect PX4 holybro(Timeout waiting for bytes, Invalid sync response, can't see setting page)

optimize0107, 2020-03-05:
## Expected Behavior
Connect PX4 holybro via QGC.

## Current Behavior
**Case 1. Can't upgrade firmware (TImeout waiting for bytes)**
[image]
The 'ok' button on the Firmware setup disappears within three seconds of the usb connection, and I can only see "upgrade cancelled".

**Case 2. Can't upgrade firmware (invalid sync respons) - same with #8213** 
same with number 1 case, but error message is changed to "Error: Sync: Send Command: Invalid sync response: 0xfe 0x3c"

**Case 3. Can't see other setup tab except 'Summary', 'Firmware' even after Firmware upgrade**
There are no tabs other than 'Summary' and 'Firmware' even if firmware can be upgraded fortunately

## Steps to Reproduce:
1. Just connect vis USB Cable

I can't tell exactly what situations become Case 1 or Case 2. The error message changes even I just pull out and plug in the USB cable. In some cases, I could upload Firmware without an error, but no matter how long I wait after Firmware is uploaded, I can't see other setting tabs except 'summary', 'Firmware' like Case 3.

## System Information
I tried with these setting.
1. OS : Window 10, Ubuntu 18.04
2. QGC version : v4.0.1, v3.5.6, v3.5.4, v3.4.3, v3.4.0
3. FC : 3 x Pixhawk4 holybro (2 x new one, 1 x used)
4. Pixhawk version : v1.10.1(stable), v1.9.2(stable), v1.9.1(stable), v1.8.0(stable), v1.7.0(stable)
5. Computer : 3 x computer, 1 x notebook, 
6. Cable : 4 x usb cable(3 x default one in the pixhawk box, 1 x my personal usb cable)
                 (no usb hub, and all usb port was USB 3.1 or USB 3.0)

## Detailed Description
Recently, px4 used in my drones have been connected by only 'mavros' with companion computer. Today I used usb cable to make two drones more, but all of the variuous situations presented above failed. And I found that the one that was connected by mavros could not be connected through usb either. 

Please help me. I can't find solution in other issues

DonLakeFlyer, 2020-03-05:
This usually means a bad usb cable. Use a different USB cable and don't connect it through a USB hub.

optimize0107, 2020-03-06:
> This usually means a bad usb cable. Use a different USB cable and don't connect it through a USB hub.

I already used four cables and three more different cables after your comment, but it didn't work. 
So I'm using telemetry now

closed 2020-03-06 by optimize0107

melisab, 2021-11-27:
@optimize0107 May I know how did you fix this issue?

czp0204, 2023-03-06:
> @optimize0107 May I know how did you fix this issue?

hi,may I know how did you fix this issue