mavlink/qgroundcontrol #8504: UI: Show relative/absolute values on Terrain information panel at appropriate scale

Saijin-Naib, 2020-03-06:
Tell us a bit about the feature:
- What problem does it solve?
  - Validation and information density
- What does it do?
  - When scaled large enough, show rel/abs values on Terrain Information panel to help validate heights of waypoints, mission items, survey plan, etc.
- What flight stacks must it work with? (All, PX4, ArduPilot)
  - All
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
  - All
- Are there other systems that have this feature?
  - No
- What communication/integration standards does the feature rely on (e.g. MAVLink commands etc.)
  - N/A
- Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)
  - If sized large enough, showing just axis labels for min/max height would be incredible (0....130ft AGL), if sized even larger, showing heights per mission item on the panel would be even better so you can see if the heights vary per item appopriately. A click/tap to toggle between absolute and relative values would also be invaluable.

DonLakeFlyer, 2020-03-06:
You should put all of these requests into a single issue for discussion. Since they are all related to the same thing which is to be able to validate a plan for correctness.

Saijin-Naib, 2020-03-06:
> You should put all of these requests into a single issue for discussion. Since they are all related to the same thing which is to be able to validate a plan for correctness.

Ah, I was under the impression that having them split out made it easier to decide yes/no and milestones/etc per particular feature request, as well as to keep discussion focused per each.

Shall I consolidate/merge them all then?

DonLakeFlyer, 2020-03-06:
Generally separate is the way to go. But this is really a discussion around what is needed to validate a Plan correctly. The key is coming at it from a different direction. I want to know the problem you are trying to solve. Giving possible solutions to it is nice, but that should be separate from what is the problem you are trying to solve. You have to be careful to not dive into solutions for the problem too fast. Otherwise you might head down the wrong road. Best to start  with the problem statement. You also have to be very careful to not jade yourself with existing implementations. That can also lead you to a non-optimal solution.

DonLakeFlyer, 2020-03-06:
> Shall I consolidate/merge them all then?

Yes. And the issue should be labeled "Plan Validation" or something like that

closed 2020-03-06 by Saijin-Naib