mavlink/qgroundcontrol #8508: Plan View has buggy first waypoints in 4.0

Mapleguy, 2020-03-06:
## Expected Behavior
When dragging or editing the first waypoint in a mission, the mission data should update. The first mission item should also be visually connected to the mission start.

## Current Behavior
Mission data is not updated in the plan view when editing or dragging the first waypoint. The waypoint is also not visually connected to the mission start.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Create a mission in the plan view
2. Add one or more waypoints
3. Drag or otherwise edit the first waypoint
4. Drag

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2020-03-06:
> The first mission item should also be visually connected to the mission start.

That is not correct with 4.0. If you do not have a Takeoff item in the Plan then the mission is assumed to start from the air. In this case it will not connect to Launch.

DonLakeFlyer, 2020-03-06:
> When dragging or editing the first waypoint in a mission, the mission data should update immediately.

Since you have no Takeoff item. The mission does not connect back to Launch. Hence there are not stats to show since there is no previous item.

DonLakeFlyer, 2020-03-06:
The other change you may notice is that QGC no longer automagically adds a Takeoff item instead of a waypoint as the first item. You now need to manage that yourself. The reason being QGC doesn't know where you want your mission to start. At this point I"m interested in feedback on all of this since it's new. It's possible that a setting for "All missions start from ground" which defaults to true may be something that would be helpful. It's unclear to me whether missions starting from the air are a 0.1% use case or more prominent.

DonLakeFlyer, 2020-03-06:
> Since you have no Takeoff item. The mission does not connect back to Launch. Hence there are no stats to show since there is no previous item.

FYI: QGC has been like this pre 4.0. Most people probably never noticed since takeoff was added automatically.

Antiheavy, 2020-03-06:
> It's unclear to me whether missions starting from the air are a 0.1% use case or more prominent.

In my experience starting a mission from the air is commonly done during system development and basically never done in a production/commercial system.  I might image some exceptions to that rule in film/video applications maybe, but I have less experience there.

Mapleguy, 2020-03-07:
In my particular case, it doesn't matter what waypoint are called or what commands are attached to them. I just need them all connected with the launch being the first point. The plan view is being used purely for planning, which needs to be simple, the actual drone doesn't use them for anything. I liked that just adding a waypoint added a takeoff as well, and I'd either like a feature to enable this or a direction to changing it back for my own uses. That would be great.

DonLakeFlyer, 2020-03-07:
I've discussed with some other folks as well and I"m going to add a setting for starting from the air.

closed 2020-03-08 by DonLakeFlyer