mavlink/qgroundcontrol #8510: UI/UX: Mission Plan Validation & Assessment

Saijin-Naib, 2020-03-06:
**Prior Feature Requests:**  
[UI: Expandable Terrain Information Panel #8505 ](https://github.com/mavlink/qgroundcontrol/issues/8505)
- Allow for the resizing of the Terrain Information panel to allow for better inspection of data, as well as possibly for dynamic changes in displayed information density

[UI: Show relative/absolute values on Terrain information panel at appropriate scale #8504](https://github.com/mavlink/qgroundcontrol/issues/8504)
- When scaled large enough, show rel/abs values on Terrain Information panel to help validate heights of waypoints, mission items, survey plan, etc.

[UI: Show both relative/absolute terrain/waypoint height values #8503](https://github.com/mavlink/qgroundcontrol/issues/8503)
- Shows both relative (set in the options, for instance) and absolute height values for appropriate plan items. So, if you setup a survey for 130ft above terrain, you'd expect to see the entire survey/waypoint list at 130ft rel, but the absolute values might vary, especially in terrain that has a lot of dynamism.

[UI: Terrain details on hover/hold #8501 ](https://github.com/mavlink/qgroundcontrol/issues/8501)
- Provides terrain height values when requested (hover on Linux/Mac/Windows, hold on Android/iOS)

[UI: Toggle Panel to show details of waypoints #8500 ](https://github.com/mavlink/qgroundcontrol/issues/8500)
- Toggle-able panel that shows details for each aspect/waypoint of a plan (command, location (lat/long), altitude (rel/abs), grade (from previous), angle (from previous), distance (from previous), and azimuth (from home).

[UX: Allow usage of sub-folders under Mission (Android/iOS) #8499](https://github.com/mavlink/qgroundcontrol/issues/8499)
- Provides an intuitive/hierarchial means to organize plans

[UI: Show Name of Saved/Loaded Plan #8498 ](https://github.com/mavlink/qgroundcontrol/issues/8498)
- Shows the plan name in the program title bar (Win/Mac/Linux), or in the window status panel (Android/iOS)

**Problem Statement:**
Verification is the primary function. Did the import work? Was the data shifted? Precision issues? CRS issues?

Can I trust this plan?  

How do I ensure that the plan is correct, site-accurate, and safe to deploy when I get to the field? I'm trusting my plan and the sUAS to do exactly what I told it that it should via the plan so I can do my job as RPIC and focus on the safety of the environment.

That's all a lot easier if I don't have "shit, was that waypoint 130ft above home, mission default height, terrain... Did the units get botched on import/export (Tower is guilty of this 🙄 [130ft imported as/into m so more like 430ft waypoint]), did I have my terrian precision set correctly so the hillock got recognized", etc.

I may be more... fussy about this than most because I'm a GIS Analyst, so these are the kinds of details I obsess about in my daily work and am very, very accustomed to not only knowing at a glance, but being in intimate control over.

**Example of site location that prompted above musing/obsessing:**
[image]

**Other Planning Software Used:**

- Mission Planner

- APM Planner2

- Solex

- Tower

- QGIS

- Google Earth Pro

DonLakeFlyer, 2020-03-06:
So @sanderux and I just had a loing discussion around this yesterday. He's going to jump in here as well.

Saijin-Naib, 2020-03-06:
Sounds great!

Antiheavy, 2020-03-06:
> Can I trust this plan?

This is the fundamental issue.  It should be possible to ensure trust without requiring the user to be an "analyst" of any type.  Lots of ways to approach this.

QGC does let you "Save Mission Waypoints as KML..." if you want to bring the mission plan into a GIS tool and look at the entire flight path in 2D/3D.

Saijin-Naib, 2020-03-06:
> QGC does let you "Save Mission Waypoints as KML..." if you want to bring the mission plan into a GIS tool and look at the entire flight path in 2D/3D.

Unfortunately, aside from KML not being my choice of interchange format for a number of reasons, the KML export isn't terribly useful in and of itself, does not retain any of the significant data of the plan, and upon import back to QGC is even less useful, preventing usage of the GIS software for any actual verification and adjustment.

[image]

sanderux, 2020-03-06:
Hi Guys,

For some time now we have been working on exactly this for our DeltaQuad clients. The system we developed is https://validator.deltaquad.com/

This tool analyzes QGC missions plans to validate a number of things such as;
- Mission sanity (takeoff, landing, max distance, etc)
- Terrain elevation (the whole path, not just the waypoints)
- Weather condition in accordance with tolerances
- Giving tips about turn-around distances etc
- Keeping an eye on max climb angles
- Making sure correct spacing is applied to takeoff and landing items
- Making sure things are flown withing safety margins
- Etc

A sample output of the result you can see here:
https://validator.deltaquad.com/validator/3d.php?uid=5e62d3536b9d75e62d3536ba26&flightdate=1583528400&localFlightdate=1583532000

I'm currently working with @DonLakeFlyer to integrate external API's such as these into QGC. I would welcome other input and see if we can co-create a validation system that is working for other types of vehicles too.

Saijin-Naib, 2020-03-06:
And what are the costs to use these services?

Antiheavy, 2020-03-07:
@sanderux super cool!

> Mission sanity (takeoff, landing, max distance, etc)

recently I've been wondering if there should be a way for QGC to save vehicle profiles that the user could select or QGC could auto-select based on download vehicle ID info.  The user/integrator could enter in a bunch of values like these and save a performance/safety profile for a certain vehicle/type.  This way QGC can keep mission planning within limits and the user can just inherently "trust" a bunch of things.

> Terrain elevation (the whole path, not just the waypoints)

yes, the whole path!  Systems I've worked with in the past have had altitude and width buffers applied to the flight path to check for terrain collisions with a safety buffer built in.

> Weather condition in accordance with tolerances

are there good free/reliable weather sources?  Or if paid, maybe allow user to enter a key, similar to mapbox integration now?

sanderux, 2020-03-07:
> And what are the costs to use these services?

We dont realy work that way. The system is currently only suitable for the deltaquad. It would be a team effort to make something good and open. If you cannot contribute in that way then PM me and i can see if i can set you up with someone that can move it forward for you

sanderux, 2020-03-07:
> @sanderux super cool!
>
>
> recently I've been wondering if there should be a way for QGC to save vehicle profiles that the user could select or QGC could auto-select based on download vehicle ID info. [...]

Indeed i am talking to don about this too. I think all defaults should be vehicle controlled.

> 
> > Terrain elevation (the whole path, not just the waypoints)
> 
> yes, the whole path! Systems I've worked with in the past have had altitude and width buffers applied to the flight path to check for terrain collisions with a safety buffer built in.

Buffer can get tricky, sometimes people want to fly close to something. But alerting them would probably be good
> 
> > Weather condition in accordance with tolerances
> 
> are there good free/reliable weather sources? Or if paid, maybe allow user to enter a key, similar to mapbox integration now?

There are a few free systems such as openweathermap which we now use

Saijin-Naib, 2020-03-07:
@sanderux , I can't program very well at all, so direct code contributions from me are unlikely, unfortunately.

closed 2026-10-04 by github-actions[bot]