mavlink/qgroundcontrol #8528: ADS-B module not working

ajinkyaavy, 2020-03-09:
## Description
I am trying to integrate an ASD-B plug-and-play module with Pixhawk4. I set up the parameters as mentioned in the documentation [here](https://docs.px4.io/master/en/advanced_features/traffic_avoidance_adsb.html). I am using the latest stable PX4 build on QGroundControl v4.0.1

## Expected behavior
Since this is a plug-and-play module, I was expecting it to start showing airplane frames as soon as I connect it. This works in MissionPlanner (flashed the Pixhawk4 with Ardupilot). 
However, when I run it on QGroundControl it doesn't work. 
I tried with both Ardupilot and PX4.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04.4 LTS
- QGC Version: 4.0.1
- QGC build: daily
- Flight Controller: Pixhawk 4
- Intended Autopilot to use: PX4

[image]

DonLakeFlyer, 2020-03-09:
First step is to use the Mavlink Inspector (analyze view, last toolbar button) to see if the vehicle is sending the ADSB_VEHICLE telemetry message. 

If aren't getting ADSB_VEHICLE messages then there is a vehicle/firmware side setup problem. YOu'll have to consult more with the firmware folks to help out.

If you can see the ADSB_VEHICLE message in the inspector then there is a QGC side bug. In this case turn on telemetry logging even when not armed. Settings/General page. Then connect you vehicle, wait until you see some ADSB_VEHICLE messages in the mavlink inspector. Disconnect the vehicle. Put the telemetry log someone where you can link to it here.

BishalSingha, 2020-03-09:
I am trying to introduce a custom set of icons for the waypoints. Any ideas on how to do that?

DonLakeFlyer, 2020-03-09:
@BishalSingha Please don't add random questions which have nothing to do with this issue. GitHub Issues are for bugs and feature requests.

ajinkyaavy, 2020-03-10:
> First step is to use the Mavlink Inspector (analyze view, last toolbar button) to see if the vehicle is sending the ADSB_VEHICLE telemetry message.
>
> If aren't getting ADSB_VEHICLE messages then [...]

Thanks for your reply. I wasn't getting ADSB_VEHICLE messages. I discussed with the company and now it works for me. I have made a small documentation of the process I followed, for future reference.

## Steps followed

- Connect the ADSB module to TELEM2 with the Pixhawk powered off.

- Power up the Pixhawk. 

- There are two LEDs on the Aerobits AERO ADS-B module (white and green). As soon as you give power to the Pixhawk, both of them will be ON. The white LED is ON only during the bootloader stage and later it will blink if a plane frame is received. The green LED will be constantly ON. This may take 5-10 seconds.

- Run QgroundControl.

- **Use the following parameter settings:**

MAV_1_CONFIG= TELEM 2
MAV_1_FORWARD= 1 
MAV_1_MODE = Normal
MAV_1_RATE= 0 B/s
SER_TEL2_BAUD= 115200

- Reboot the vehicle.

closed 2020-03-10 by ajinkyaavy