mavlink/qgroundcontrol #8533: 4.0.2 can't run on Ubuntu 16.04: version `GLIBC_2.27' not found

bys1123, 2020-03-10:
Is QGC 4.0 not support 16.04 now?

maowen, 2020-03-10:
I am also running Ubuntu 16.04 and can't run . This is the error I get when running the latest release, v4.0.2, of QGC. I can still successfully run QGC v3.5.6

    $ ./QGroundControl.AppImage 
    /tmp/.mount_i4hPuB/QGroundControl: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /tmp/.mount_i4hPuB/QGroundControl)

DonLakeFlyer, 2020-03-10:
I don't think so. @andrewvoznytsa Is that correct?

DonLakeFlyer, 2020-03-10:
Dup of #8438

andrewvoznytsa, 2020-03-11:
That error is absolutely correct - 16.04 supports glibc 2.23 only (see https://launchpad.net/ubuntu/xenial/+source/glibc).

We are building QGC on 18.04 which is using 2.27 and this information is recorded in application executable. There are a few ways to solve that situation - build on 16.04 (safe) or trick linker, forcing it to record 2.23 reference (last resort solution, very dangerous).
Building on 16.04 is not very practical in generic case since it is missing one of gstreamer packages - we'll end up without video support. Moreover - we have some bug that needs migration to 20.04 due to gstreamer version (we need to migrate to gstreamer 1.16+ or ship it ourselves which is also ....).

I think the most realistic solution for 16.04 is to provide QGC without video - porting gstreamer back to 16.04 does not sound right.

laxmanbobby, 2020-03-31:
I can confirm this error on Ubuntu 16.04. 

What was the solution?

DonLakeFlyer, 2020-03-31:
The only solution is to build QGC yourself without video support. Starting with 4.0 the official dropped versions of QGC no longer support Ubuntu 16.04.

@hamishwillee This should go into the docs somewhere.

closed 2020-03-31 by DonLakeFlyer

laxmanbobby, 2020-03-31:
@DonLakeFlyer 

Could you please provide a reference on how to do this? Your work is greatly appreciated.

DonLakeFlyer, 2020-03-31:
Documentation is your friend: http://qgroundcontrol.com., https://dev.qgroundcontrol.com/en/getting_started/

hamishwillee, 2020-04-01:
Fixed in https://github.com/mavlink/qgc-user-guide/pull/261 - @DonLakeFlyer requies your review.

DonLakeFlyer, 2020-04-01:
Thanks @hamishwillee

akshata-1, 2021-11-23:
Hey did you guys solve this?
Ubuntu version- 18.04 LTS 
Error
/tmp/.mount_QGrountAgtKr/QGroundControl: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_QGrountAgtKr/QGroundControl)
/tmp/.mount_QGrountAgtKr/QGroundControl: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_QGrountAgtKr/QGroundControl)