mavlink/qgroundcontrol #8535: Lubuntu Qt error (core dumped)

albydnc, 2020-03-10:
On lubuntu 18.04 LTS, after following the guide on the website, I can't get the app to launch.
here the dump: 
```
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_tWnjaN/Qt/plugins/platforms" ...
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqeglfs.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqeglfs.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "eglfs"
        ]
    },
    "archreq": 0,
    "className": "QEglFSIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("eglfs")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqlinuxfb.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqlinuxfb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "linuxfb"
        ]
    },
    "archreq": 0,
    "className": "QLinuxFbIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("linuxfb")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqminimal.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqminimal.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimal"
        ]
    },
    "archreq": 0,
    "className": "QMinimalIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("minimal")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqminimalegl.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqminimalegl.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimalegl"
        ]
    },
    "archreq": 0,
    "className": "QMinimalEglIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("minimalegl")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqoffscreen.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqoffscreen.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "offscreen"
        ]
    },
    "archreq": 0,
    "className": "QOffscreenIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("offscreen")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqvnc.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqvnc.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "vnc"
        ]
    },
    "archreq": 0,
    "className": "QVncIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("vnc")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-egl.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-egl.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "wayland-egl"
        ]
    },
    "archreq": 0,
    "className": "QWaylandEglPlatformIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("wayland-egl")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-generic.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-generic.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "wayland"
        ]
    },
    "archreq": 0,
    "className": "QWaylandIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("wayland")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-xcomposite-egl.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-xcomposite-egl.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "wayland-xcomposite-egl"
        ]
    },
    "archreq": 0,
    "className": "QWaylandXCompositeEglPlatformIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("wayland-xcomposite-egl")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-xcomposite-glx.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqwayland-xcomposite-glx.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "wayland-xcomposite-glx"
        ]
    },
    "archreq": 0,
    "className": "QWaylandXCompositeGlxPlatformIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("wayland-xcomposite-glx")
QFactoryLoader::QFactoryLoader() looking at "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqxcb.so"
Found metadata in lib /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqxcb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "xcb"
        ]
    },
    "archreq": 0,
    "className": "QXcbIntegrationPlugin",
    "debug": false,
    "version": 330752
}

Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/tmp/.mount_tWnjaN/platforms" ...
Cannot load library /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqxcb.so: (libxkbcommon-x11.so.0: cannot open shared object file: No such file or directory)
QLibraryPrivate::loadPlugin failed on "/tmp/.mount_tWnjaN/Qt/plugins/platforms/libqxcb.so" : "Cannot load library /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqxcb.so: (libxkbcommon-x11.so.0: cannot open shared object file: No such file or directory)"
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted (core dumped)
```

albydnc, 2020-03-10:
The version is the latest stable one.

DonLakeFlyer, 2020-03-10:
@dagar Do you think you could help out here?

They key info is:
```Cannot load library /tmp/.mount_tWnjaN/Qt/plugins/platforms/libqxcb.so: (libxkbcommon-x11.so.0: cannot open shared object file: No such file or directory)```

Which means one of the Qt plugins is missing something it needs. Closed search result I could get for libxkbcommon-x11.so.0 is it has something to do with non-english on Linux. No idea how to resolve this. Does it require an addition to AppImage creating scripts?

DonLakeFlyer, 2020-03-11:
According to Qt release notes:
```
 - [QTBUG-65503] Removed xkbcommon from bundled sources. This library is
   present on all supported platforms. The minimal required version now is
   0.5.0.
```

DonLakeFlyer, 2020-03-23:
I believe this is fixed now. Can you verify in latest stable?