mavlink/qgroundcontrol #8550: [Feature Request] use mobile wireless data for internet with Wi-Fi is used to connect to drone

Antiheavy, 2020-03-11:
When using a mobile device and connecting to the vehicle's comm link over Wi-Fi, certain things that use the internet (e.g. maps and terrain) don't seem smart enough to try the mobile device's wireless connection (e.g. LTE).

It is possible to let some parts of QGC use a different network connection to get to the internet when the comm link is using WiFi (or vice verse)?

DonLakeFlyer, 2020-03-12:
I don't think the OS allows this sort of thing. Not sure any OS does in reality. @dogmaphobic Do you know anything about this sort of stuff?

Antiheavy, 2020-03-19:
Some I was talking to suggested this as a possible approach: 
https://developer.apple.com/documentation/foundation/urlsessionconfiguration/improving_network_reliability_using_multipath_tcp

DonLakeFlyer, 2020-03-21:
Or something like this: https://apple.stackexchange.com/questions/173894/can-i-connect-to-wifi-but-still-use-cellular

booo, 2020-03-26:
The dhcp client receives a default route and a name server from the dhcp server of the wifi bridge. You can disable this behaviour on the server or on the client side. E.g. on machines with networkmanager you should be able to disable the request of a default route with `nmcli device modify ffdhcpvlan11 ipv4.never-default yes`.

closed 2026-10-04 by github-actions[bot]