mavlink/qgroundcontrol #8553: Qt Creator build Chinese localization "qgc_zh_CN.ts" into "qt_zh_TW.qm"

bys1123, 2020-03-13:
[image]
This will makes Debug console shows 'Qt localization for "zh_CN" is not present'
I tried to build "qt_zh_CN.qm" with Qt Linguist, this will get a 251kb file. The auto build "qt_zh_TW.qm" only have 125KB, seems have a different encoding.

There are also few bugs about Chinese language.

Trying to fix this.

bys1123, 2020-03-14:
[image]
All pre-build files looks no problem.

bys1123, 2020-03-14:
This a bug of qt 5.12.6

bys1123, 2020-03-15:
[image]
Tried to build a micro project, nothing goes wrong.

DonLakeFlyer, 2020-03-15:
Adding Gus...

bys1123, 2020-03-15:
[image]
Try to copy qgc localization files to new project, works fine.

bys1123, 2020-03-15:
[image]
Copy all the localization files to new project, then build, nothing wrong, but no translation folder created, and not changed the prefix.

bys1123, 2020-03-15:
[image]
Then must someting wrong in here.

bys1123, 2020-03-16:
[image]
Tried run windeployqt in my project, then I only have zh_TW now.

bys1123, 2020-03-16:
5.12.7 has the same problem

bys1123, 2020-03-16:
Tried to created a qgc_zh_TW.ts file see what will happen. Still only have qt_zh_TW.qm. Seems windeployqt didn't handle qgc_zh_CN.qm at all.

DonLakeFlyer, 2020-11-06:
I think this is all working now

closed 2020-11-06 by DonLakeFlyer

magicjiuwen, 2023-04-24:
> Tried to created a qgc_zh_TW.ts file see what will happen. Still only have qt_zh_TW.qm. Seems windeployqt didn't handle qgc_zh_CN.qm at all.

Hi @bys1123 , 我也遇到了同样的问题，请问你是怎么解决的呢？感谢不吝赐教！