mavlink/qgroundcontrol #8557: [Feature Request] 10x speed option on telemetry replay

Antiheavy, 2020-03-14:
Users often have very long telemetry logs if they have left the application up and running all day.  a 10x speed option on replay would be nice!  thanks.

[image]

DonLakeFlyer, 2020-03-16:
If I remember correctly unless you have a quantum computer anything over 5x would just hang things up in the ui. What is the goal to 10x? Are they trying to find some point in the log? How about a mechanism to search ahead for whatever they are looking for?

Antiheavy, 2020-03-16:
I tried 10x speed on a custom build and it seemed to work on my 6 yr old windows PC. I didn’t test it exhaustively though. I’ll do more tests on different tlogs with different UI items going (e.g. analysis). 

Sometimes the goal is to replay the whole log without jumping /skipping around. E.g to see the whole ground track on the map or to watch some telemetry analysis charts go the whole flight without skipping.

Antiheavy, 2020-11-12:
> If I remember correctly unless you have a quantum computer anything over 5x would just hang things up in the ui.

@DonLakeFlyer you might try this again in you are curious.  I've been using 10x replay speed on 100's of log replays the past year and it works fine.

booo, 2022-09-15:
As @Antiheavy reported the changed worked quite well for him we could implement this with a simple patch I guess:

```diff
diff --git a/src/QmlControls/LogReplayStatusBar.qml b/src/QmlControls/LogReplayStatusBar.qml
index a1733c5c8..7bc3bfd4c 100644
--- a/src/QmlControls/LogReplayStatusBar.qml
+++ b/src/QmlControls/LogReplayStatusBar.qml
@@ -69,6 +69,7 @@ Rectangle {
                 ListElement { text: "1x";   value: 1 }
                 ListElement { text: "2x";   value: 2 }
                 ListElement { text: "5x";   value: 5 }
+                ListElement { text: "10x";  value: 10 }
             }
 
             onActivated: controller.playbackSpeed = model.get(currentIndex).value
```

closed 2024-09-22 by HTRamsey