mavlink/qgroundcontrol #8558: Can't select waypoints above mission item 256

Antiheavy, 2020-03-14:
User reported being unable to select waypoints with a mission index >256.  Attempting to do so would cause the vehicle to fly back to the start of the mission.  User sent me their .plan file and I can see a difference in selecting animations for waypoints 256 and below compared to waypoints >256.

I've been able to recreate this with other large survey missions with >256 mission items.

Here is a short video showing what I see on the ground:
https://youtu.be/H9Oysp6oGZo

Antiheavy, 2020-03-15:
Looks like the waypoint index sometimes rolls over.  e.g. selecting 260 might roll over and actually select 4.

DonLakeFlyer, 2020-03-15:
Just shows how many people do this with large numbers of waypoints. This bug has been there for ages.

Antiheavy, 2020-03-15:
https://github.com/mavlink/qgroundcontrol/pull/8566  fixes the issue in our build!

closed 2020-03-15 by Antiheavy