mavlink/qgroundcontrol #8571: [Bug] UI Error, unable to bind spektrum receivers after QGC v4.0.0

SwiftGust, 2020-03-17:
## Expected Behavior
Please tell us what should happen as detailed as possible.
QGC should be able to bind spektrum receiver.

## Current Behavior
Please describe what happens instead of the expected behavior.
QGC cannot bind to spektrum radio,
Ok button in the screenshot doesn't respond and displays the following message.
`qrc:/qml/RadioComponent.qml: ReferenceError: radioGroup is not defined`
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect to any vehicle
2. Go to Radio pane
3. Click Spektrum Bind button
4. Select any of receiver types and press Ok
5. You will get `qrc:/qml/RadioComponent.qml: ReferenceError: radioGroup is not defined`

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04
- QGC Version: QGC v4.0.0
- QGC build: stable release
- Autopilot (with version): PX4 1.9.0 dev

DonLakeFlyer, 2020-03-18:
@SwiftGust Could you verify in a daily build from tomorrow that this is working for you now?

closed 2020-03-18 by DonLakeFlyer

SwiftGust, 2020-03-18:
@DonLakeFlyer I just tested spektrum bind and it's working well. Thank you.

DonLakeFlyer, 2020-03-18:
@SwiftGust Thanks so much for reporting the problem and testing the fix

PR #8573 (2020-03-18): Fix Radio button problems
Fixes: #8571, #8572

Also reworked Copter RTL settings to make more sense:
[image]