mavlink/qgroundcontrol #8572: APMSafteyComponent Return To Launch Radio Buttons Incorrect Exclusivity

almatthew, 2020-03-17:
## Expected Behavior
The "Return at current altitude" and "Return at specified altitude" radio buttons should not be exclusive to "Land with descent speed:" and "Final loiter altitude".

The user should be able to specify "Land with descent speed" and "Final loiter altitude" independent of other parameters within the "Return to launch" section.

## Current Behavior
Clicking Return at current altitude", "Return at specified altitude", "Land with descent speed:", "Final loiter altitude" are all mutually exclusive.  RTL altitude, descent speed and final altitude text fields are enabled/disabled based on these radio buttons.

## Steps to Reproduce:
While using APM, in the Safety page:
1. Click "Return at current altitude"
[image]
2. Click "Return at specified altitude"
[image]
3. Click "Land with descent speed:"
[image]
4. Click "Final loiter altitude"
[image]

## System Information
When posting bug reports, include the following information
- Operating System: All supported operating systems
- QGC Version: 4.0 or later, but any version after April 29, 2019
- QGC build: All
- Flight Controller: APM SITL, PixhawkCube, PixRacer and all other APM-supported FCs.
- Autopilot (with version): ArduCopter Copter 3.6 and later (and likely earlier)

## Detailed Description
This issue appears to have been introduced in April 2019: https://github.com/mavlink/qgroundcontrol/commit/182b76afcc831484fd61fc43d27160e96d50b33e#diff-bb6903a0ce2a65eaeea61c0ecea28d5d

Perhaps a better configuration for this section of the Safety page would be to eliminate the radio buttons for "Land with descent speed" and "Final loiter altitude".  Instead, always enable the "Land with descent speed" `FactTextField` and replace "Final loiter altitude" with a checkbox.

DonLakeFlyer, 2020-03-17:
Thanks for the report. This is certainly screwy. RTL_ALT_FINAL is a bit confusing. It says "This is the altitude the vehicle will move to as the final stage of Returning to Launch or after completing a mission.  Set to zero to land." Isn't that last "Set to zero to land" somewhat incorrect or at best confusing? Not sure a checkbox there will be understandable. I think RTL_ALT_FINAL =0 would mean, there is no final phase and LAND_SPEEd would never be used. Seems like better to just make both just be simple entry fields. For people that understand what RTL_ALT_FINAL=0 actually does then they can still do it.

DonLakeFlyer, 2020-03-17:
Also the commit is showing me that other radio buttons were broken by this change as well.

DonLakeFlyer, 2020-03-17:
@almatthew How about this:
[image]

almatthew, 2020-03-17:
Hi @DonLakeFlyer, thank you for the prompt response!  I agree that the two text fields make sense, and that a checkbox is not necessary. The text descriptions of those boxes might be a bit confusing to users (maybe not), but I think an APM dev should probably chime in here if possible (@rmackay9 @tridge @peterbarker)?

closed 2026-10-05 by github-actions[bot]