mavlink/qgroundcontrol #8578: gimbal yaw indicator erroneously propagates to followup waypoints

remster, 2020-03-20:
## Expected Behavior
When the gimbal checkbox isn't checked on a waypoint - the [visual](https://github.com/mavlink/qgroundcontrol/blob/master/src/QmlControls/MissionItemIndexLabel.qml#L60) shouldn't show.

## Current Behavior
it shows
[image]

## Steps to Reproduce:
1. create a plan with a number of waypoints
2. check camera->gimbal on the Nth
3. all waypoints that follow Nth show the same gimbal heading/visual

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04
- QGC Version: a22c17b455a5cca6018d0f5abed4b3c2537b7a18
- QGC build: off master 
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
I believe this is how to fix it:
```
--- a/src/MissionManager/MissionController.cc
+++ b/src/MissionManager/MissionController.cc
@@ -1333,10 +1333,14 @@ void MissionController::_recalcMissionFlightStatus()
         double gimbalYaw = item->specifiedGimbalYaw();
         if (!qIsNaN(gimbalYaw)) {
             _missionFlightStatus.gimbalYaw = gimbalYaw;
+        } else {
+            _missionFlightStatus.gimbalYaw = qQNaN();
         }
         double gimbalPitch = item->specifiedGimbalPitch();
         if (!qIsNaN(gimbalPitch)) {
             _missionFlightStatus.gimbalPitch = gimbalPitch;
+        } else {
+            _missionFlightStatus.gimbalPitch = qQNaN();
         }
```

DonLakeFlyer, 2020-03-23:
If you adjust the gimbal, that adjustment continues on until you make another adjustment. Leaving the gimbal settings unchecked does not bring the gimbal back to neutral they stay where they were before. Hence the visual continue to show.

remster, 2020-03-24:
Thanks. Is there a purpose (feature/use-case) to this behaviour?
Is there a purpose in setting the gimbal pitch for the whole mission (via the `SettingsItem`) - behaviour protected by [this](https://github.com/mavlink/qgroundcontrol/blob/master/src/MissionManager/MissionControllerTest.cc#L169) test case?
Is it possible to set it back to `NaN` (I can't see how there is)?
Or that it starts maintaining the heading towards the next waypoint?

DonLakeFlyer, 2020-03-24:
The fact that if you change the gimbal angle it stays that way unless you change it back is just the way missions work. QGC hides some of this from you in the way it adds gimbal change support to a waypoint item. Behind the scenes this expands out to multiple command: WP, GIMBAL change.

> Is it possible to set it back to `NaN` (I can't see how there is)?
> Or that it starts maintaining the heading towards the next waypoint?

That is a great question and I don't know the answer to that. From looking at the command I'm not sure it's possible. I'll ask the mavlink folks about that.

DonLakeFlyer, 2020-03-24:
> That is a great question and I don't know the answer to that.

No wait. Since change gimbal angles are just that (you're aren't targetting a specific ROI) you just set them back to 0 don't you?

remster, 2020-03-24:
So I am interpreting you correctly: gimbal angles are just angles - i.e.: the physical gimbal is always at an angle(s) (it's never physically NaN). So all QGC does interpreting NaN is visualising that it doesn't know that angle.

[image]

It only knows it after the user gives it some specific value (at that bolded/dark green waypoint) and it keeps that information for the followup watpoints. Is that the philosophy? 
If it is - it is coherent - but I don't see it very useful.

I would like to to propose that the UI shows the gimbal when the user expresses interest in it (sets it to something) - just to give them a cue as to what they did and separate this from what's an accident of previous value and the drone's heading. I assume it's hardly ever useful to re-use the gimbal **yaw** waypoint to waypoint (it is prolly quite different to gimbal pitch).

I a proposing because this appears a simple change that I could actually do.

DonLakeFlyer, 2020-03-25:
> So all QGC does interpreting NaN is visualising that it doesn't know that angle.

Not quite sure what you mean here. I think you re referring to an internal nuance of the way this is coded. 

> but I don't see it very useful.

That may be the case but there are commercial vehicle operators out there who use QGC and expect it to work this way visually.

remster, 2020-03-25:
> > So all QGC does interpreting NaN is visualising that it doesn't know that angle.
> 
> Not quite sure what you mean here. I think you re referring to an internal nuance of the way this is coded.

I am trying to find logic in why the gimbal yaw **isn't** show to the left of the bolded / dark green waypoints:
[image]
I am guessing - that logic is that gimbal yaw isn't known to QGC - which is why it isn't showing it up to the first waypoint (dark green) where it is known.
> 
> > but I don't see it very useful.
> 
> That may be the case but there are commercial vehicle operators out there who use QGC and expect it to work this way visually.

Other than - i am genuinely curious why do they expect it to work this way - if you think they do - you can close this ticket.

DonLakeFlyer, 2020-03-25:
> I am trying to find logic in why the gimbal yaw **isn't** shown to the left of the bolded / dark green waypoint

That is certainly a valid question. The goal from the start was to not show gimbal indicators unless someone actually changed the gimbal yaw. This prevents the display from becoming too noisy for for users who do nothing with gimbals. But that was mostly for the case of the gimbal is never changed. When the gimbal is changed mid-flight I do think there may be some value in seeing the point where you actually told the gimbal to start doing something different from what it would normally do. Hence the no indicators prior to the change. But in reality that is not a super strong argument. It may likely be better to show them throughout the flight path if you ever change the gimbal. Seems kind of 50/50 choice to me. Not sure what is better.

remster, 2020-04-17:
As I have a need to show it only on waypoints where it's explicitly set - i could volunteer a change that puts this dilemma behind a setting. How would that sound?

closed 2020-05-28 by remster