mavlink/qgroundcontrol #8580: gimbal heading indicator seemingly off-by-one

remster, 2020-03-20:
## Expected Behavior
The gimbal [visual](https://github.com/mavlink/qgroundcontrol/blob/master/src/QmlControls/MissionItemIndexLabel.qml#L60) should match the heading + gimbal yaw.

## Current Behavior
It doesn't and is seemingly off-by-one.
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Unzip and load the [bad_heading.zip](https://github.com/mavlink/qgroundcontrol/files/4361370/bad_heading.zip) `.plan`

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04
- QGC Version: a22c17b455a5cca6018d0f5abed4b3c2537b7a18
- QGC build: off master 
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
The attached and screen-shut plan goes south to north in a straight line. In between the start and end there are three waypoints with heading looking (south to north):
```
                "command": 16,
                "doJumpId": 3,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    88.60905140566709, //east
...
                "command": 16,
                "doJumpId": 7,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    268.6093254982461, //west
...
                "command": 16,
                "doJumpId": 11,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    88.6090561150852, //east
``` 
the visuals show west, east, east and if one keeps on adding more waypoints they suggest it's an off-by-one problem

DonLakeFlyer, 2020-03-21:
What version of QGC is this. I thought I fixed the off by one problem recently.

remster, 2020-03-21:
I am pretty sure, yesterday i have pulled, built and reproed the defect on: 
```
commit a22c17b455a5cca6018d0f5abed4b3c2537b7a18 (HEAD -> master, origin/master, origin/HEAD)
Author: PX4BuildBot <bot@px4.io>
Date:   Fri Mar 20 10:42:04 2020 +0000

    Update PX4 Firmware metadata Fri Mar 20 10:42:04 UTC 2020

```

closed 2026-10-05 by github-actions[bot]