mavlink/qgroundcontrol #8585: "Set RC to Param..." button no action 4.0.1

TheSoundMan000, 2020-03-23:
Windows 10
4.0.1 Downloaded direct from website
Pixhawk 4/Pixhawk 2, both latest stable PX4

Drone configured and maidened with obvious PID tuning necessary. PARAM1 and PARAM2 assigned to channel 11 and 12 respectively (analog knobs). Select a desired paramater eg MC_ROLLRATE_P and advanced then "Set RC to Param..." and nothing happens.

Completely unable to assign RC control to tune the PIDs mid flight.

DonLakeFlyer, 2020-03-23:
@dogmaphobic This is sitll missing from Stable

TheSoundMan000, 2020-03-23:
Is it available in the latest daily? I might give that a burn if so. Thanks for your help guys.

DonLakeFlyer, 2020-03-23:
It's not available anywhere

TheSoundMan000, 2020-03-23:
Ah right... so what version is depicted in the above screen shot? That's straight from the PX4 1.9.0 manual. I don't mind if its beta or anything like that just chasing an implementation of some form. Is this a new feature or an obsolete one or whats the situation?

DonLakeFlyer, 2020-03-24:
The is QGC v3.5.6, it is missing from newer versions. You can get older version from the GitHub releases page.

PR #8657 (2020-04-14): Set RC To Param support
* Fixes: #8585, #8382
* Also includes some changes to showPopupDialog:
  * No need to pass in title and buttons. You now set them in the component itself.
  * Ability to pass additional dialog properties from the showPopupDialog call. This allow you to provide external context to the dialog if needed.
closed 2020-04-14 by DonLakeFlyer