mavlink/qgroundcontrol #8592: App crash when using Plan Wisard: Plan -> Create Plan -> Survey

appetito, 2020-03-24:
## Expected Behavior
Starts creating survey plan with wisard
## Current Behavior
Segmentation fault

## Steps to Reproduce:
1. Select Plan view
2. Select Survey option in Create Plan list

## System Information
When posting bug reports, include the following information
- Operating System: MacOS X
- QGC Version: 4.0.0, 4.0.4
- QGC build: stable
- Flight Controller: not connected, radionink mini pix
- Autopilot (with version): not connected, arduplane v 4.0.1

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
macos conole log:
MissionControllerLog: _addMissionSettings
MissionControllerLog: _recalcWaypointLines homePositionValid false
MissionControllerLog: _recalcMissionFlightStatus
MissionControllerLog: _recalcWaypointLines homePositionValid false
MissionControllerLog: _recalcMissionFlightStatus
MissionControllerLog: _recalcWaypointLines homePositionValid true
MissionControllerLog: _recalcMissionFlightStatus
MissionControllerLog: _recalcWaypointLines homePositionValid true
MissionControllerLog: _recalcMissionFlightStatus
MissionControllerLog: _recalcWaypointLines homePositionValid true
Segmentation fault: 11

DonLakeFlyer, 2020-03-24:
Caused by changes made to support #8023

DonLakeFlyer, 2020-03-25:
Thanks for much for the bug report. This has been fixed and I just marked v4.0.5 to go out. Android v4.0.5 will start rolling out to store in a few minutes.

closed 2020-03-25 by DonLakeFlyer