mavlink/qgroundcontrol #8607: European Regulations Geo-Awareness

kuiskas, 2020-03-26:
Tell us a bit about the feature:

- What problem does it solve?
According to the new UAS regulation from EASA, 2019/945, it will be necessary to include:
1. MobilityandTransportGeo-Awareness –no changeInterface to load and update data containing information on airspace limitations;
2. Warning alert when a potential breach of airspace limitations is detected.
3. Information on the UA’s status as well as a warning alert when its positioning or navigation cannot ensure the proper functioning of the system

- What flight stacks must it work with? (All, PX4, ArduPilot)
All
- Relevant vehicle types? (All, multirotor, fixed-wing, VTOL, submarine, etc.)
Vehicles under the class C1, C2 or C3. 
- Are there other systems that have this feature?
Dji already provide this services. 

Is there any plan to integrate this feature on QGroundControl?

choel83, 2021-01-11:
Since the new regulations are here I'll *bump* this post.

albert21martins, 2021-01-14:
Thank you @choel83. Hope some one of the team can answer,.

albert21martins, 2021-01-14:
Thank you @choel83. Hope some one of the team can answer.

kuiskas, 2021-01-14:
Thank you @choel83. Hope some one of the team can answer,.

closed 2026-10-05 by github-actions[bot]