mavlink/qgroundcontrol #8609: error: C2259: ‘CustomVideoReceiver’: cannot instantiate abstract class

NectroX, 2020-03-26:
i am having issues trying to build the custom-example..
error: C2259: ‘CustomVideoReceiver’: cannot instantiate abstract class
[image]

andrewvoznytsa, 2020-03-26:
Should be fixed by https://github.com/mavlink/qgroundcontrol/pull/8596

If you still see issue then please re-open this ticket.

closed 2020-03-26 by andrewvoznytsa

DonLakeFlyer, 2020-03-26:
Thanks @andrewvoznytsa

NectroX, 2020-03-26:
Thanks too @andrewvoznytsa
however i get this new error
[image]
is there a fix for this?

andrewvoznytsa, 2020-03-26:
Probably something broken in build systems. Will check later today.

NectroX, 2020-03-26:
> Probably something broken in build systems. Will check later today.
> […](#)

Thanks so much!

andrewvoznytsa, 2020-03-26:
> https://github.com/andrewvoznytsa however i get this new error [image:
> error c1083]

I've double checked everything - builds just fine on my side and CI. You
need to prepare proper bug report. And make sure that you follow all
READMEs if you are building for first time.

NectroX, 2020-03-27:
> I've double checked everything - builds just fine on my side and CI. You need to prepare proper bug report. And make sure that you follow all READMEs if you are building for first time.
> […](#)

Hmm that's weird..not sure what I'm doing wrong here, sorry I'm quite new to this.
I am still getting these errors
[image]

NectroX, 2020-03-27:
Is there any chance that you could provide the installation file for the custom build?

farahats9, 2020-03-28:
I am having the same issue, was anyone able to find a solution?