mavlink/qgroundcontrol #8627: Survey pane should default to the Presets view if presets exist

Antiheavy, 2020-04-01:
Using "master-12145" build from Appveyor (most recent as of time of this post) it seems that the survey pane always defaults to the Grid view, even if presets are saved.

If presets are saved it should probably default to the presets view when creating a new survey.

DonLakeFlyer, 2020-04-01:
I forgot what I had done here: I added a setting on planView: displayPresetsTabFirst which you can override to make this happen. I can't remember why I did that instead. It seems like showing first if presets exists is still better.

Antiheavy, 2020-04-01:
> I added a setting on planView: displayPresetsTabFirst which you can override to make this happen.

that must be a pluggin thing and not a general QGC setting?  I'm fine either way, but I did't see it here:
[image]

DonLakeFlyer, 2020-04-02:
It's not exposed in the UI. Only meant for custom use.

Antiheavy, 2020-04-02:
> It's not exposed in the UI. Only meant for custom use.

sounds good to me.  I'll close this issue.  If you feel like it might be valid to consider the always defaulting to preset view if presets exist then you can re-open if you want to track.

DonLakeFlyer, 2020-04-03:
I want to keep this open. It still may make sense to do this.

Antiheavy, 2020-04-04:
Just playing around here: this is a hypothetical Survey pane layout that emphasizes Presets if they exist.  All the Survey pane settings are exposed if "custom" is selected from the Presets list, including the ability to save a new preset.  If no presets exist then "custom" is the only option in the list.
[image]

Antiheavy, 2020-04-11:
We've had users report being confused by the Camera and Grid tabs.  Basically the Grid view looks totally different depending on if you have a Manual, Custom, or saved camera and there isn't an indication on that links the two views.

I wonder if going back to a single giant Survey pane would actually be less confusing?
[image]

DonLakeFlyer, 2020-12-18:
Not going to be able to get this into 4.1. Other stuff is just taking too long to deal with.