mavlink/qgroundcontrol #8628: Unable to download gps_helper.h, Qt creator fails build

laxmanbobby, 2020-04-01:
Hello, 

I am trying to build QgroundControl using Qt Creator and am following the steps described in the online guide: https://dev.qgroundcontrol.com/en/getting_started/

When I run the following command on my command line:
`git clone https://github.com/mavlink/qgroundcontrol.git --recursive`

The files begin downloading but eventually the download returns the following error: 
`fatal: reference is not a tree: 2bfc6ffb3ba74b60deab5aef8e34c2cb909c4edc
Unable to checkout '2bfc6ffb3ba74b60deab5aef8e34c2cb909c4edc' in submodule path 'src/GPS/Drivers'`

As a result, the command:
`rhb@rhb-VirtualBox:~$ git submodule update  
fatal: Not a git repository (or any of the parent directories): .git`

DonLakeFlyer, 2020-04-01:
From the text you deleted: "The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources"

closed 2020-04-01 by DonLakeFlyer

laxmanbobby, 2020-04-01:
Topic moved to: https://discuss.px4.io/t/unable-to-download-gps-helper-h-qt-creator-fails-build/16022