mavlink/qgroundcontrol #8631: UDP H.264 was way better on 3.5.6, with quality and latency has degraded significantly.

The1only, 2020-04-01:
QGC 3.5.6 vs. QGC 4.x
----

System:

MacBook Pro 2018 OSX 10.15.3
and
Acer Nitro 5, with Nvidia 1650GPU, Ubuntu 18.04

both got same issue...

## Expected Behavior
UDP H.264 without gray images or any error, very little latency.

## Current Behavior
UDP H.264 very long latency, often error in image
on the old 3.5.6. perfect....

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
Play and run...  gStreamer created video stream with speed-preset=superfast 

## Detailed Description
gst-launch-1.0 videotestsrc ! video/x-raw,framerate=20/1 ! videoscale ! videoconvert ! x264enc tune=zerolatency bitrate=3000 speed-preset=superfast ! rtph264pay ! udpsink host=10.0.0.100 port=5600

andrewvoznytsa, 2020-04-02:
I think the X at the end of `QGC 4.x` is making difference here.

If you are using 4.0.5 release then there is low latency option - please use that and confirm that what you see is not worse than 3.5.

But if you are using other 4.0 version or master then most likely this option is not available and you really can expect latency issues.

'Low latency' feature will be backported from 4.0.5 into master shortly - most likely during this week.

The1only, 2020-04-03:
3.5.6 Perfect, never any issues

4.0. latency and a loot of gray bad images

4.05 no video at all, and missing toolbar on topp of the APP

Linux Ubuntu 18.04

The1only, 2020-04-03:
PS: The quality and latency you are getting with 3.5.6 is amazing, I am not able to replicate that using simple python and gStreamer.

Make sure you study it carefully as it is absolutely prefect.

Terje

The1only, 2020-04-03:
When you compiled the binary release for 3.5.6. for Linux (Ubuntu 18.04) what version og gStreamer did you point to????
Did you use 1.14.4 ?

jaxxzer, 2020-04-08:
> 4.05 no video at all, and missing toolbar on topp of the APP

Can you double check your video settings on the _General Settings_ page? Perhaps upload a screenshot?

taileron, 2020-04-20:
For me the same, significant higher latency (about double) @ low latency setting.
And the target aspect ratio doesn´t apply - value of 1.77777 doesn´t change the 4:3 image anymore.
Part of the video is hidden by the title bar because it is no longer transparent.
And the click in the middle of the screen does not hide the elements anymore. 
Behaviour is on the other platforms e.g. Android identical.
Mac fullscreen is no more available.
I´m using low latency hardware rtsp encoders (Vivotek VS8100-V2).

andrewvoznytsa, 2020-04-26:
@taileron may I ask you to create separate ticket for aspect ratio and transparency issues? And please add some details to make it easier to understand and reproduce. For example - does your stream have PAR 1:1? What are resolutions of your screen and video? Why would you need to change aspect ratio? Also adding some screenshots to show transparency issue would be helpful.

The reason for new ticket is that:
1) these issues are no longer related to video receiver and belong to UI
2) current ticket is about latency - I'd stick with one issue/one ticket scheme

BoKoppel, 2020-05-28:
Hi!
Having similar problem with low latency mode combined with video recording.

Using video recording in QGC 4.0.5 version gives a lot of frozen frames with a grey image.
This occur on both livestream in QGC and recording. In recording there is no “grey frame effect” but frame freezes instead.
Processor load is general higher than earlier QGC 3.5.2, 30% without recording, 40-45 with recording.
Effect is intermittent, can run for minutes OK, followed by several in 30 seconds.
Always happens 6 seconds after starting recording.
When the effect occurs QGC processor load typically goes from 40 to 80%,
I thought new video processing should mean lower processor load, not higher.
Also new 64 bit QGC should be more efficient, not less?
Using internal Intel graphic card instead of Nvidia card gives same result, both in processor load and frozen frames. Maybe that could be a hint since using nVidia card in my mind should take load of processor?
Screenshot at BR: https://discuss.bluerobotics.com/t/video-freezing-and-grey-frames-in-qgc-4-0-5-when-recording/7423/2

Can I log or help to find solution in anyway?

Bo

BoKoppel, 2020-06-01:
Testing QGC 4.0.8, better but still some frozen frames seldom.
Screenshot:
https://discuss.bluerobotics.com/t/video-freezing-and-grey-frames-in-qgc-4-0-5-when-recording/7423/8

BoKoppel, 2020-06-07:
More fault tracking:

Disable the Intel onboard GPU did not work.
All graphics passes the onboard Intel, even if its processed by Nvidia GPU.
That is the normal case, even in high end laptops.
Checking gamers forums I’ve seen similar problems due to bugs in games related to handling of dual graphics.

Since this worked on 32bit QGC 3.x but not in the 64 bit QGC 4.x I guess there is an issue in graphics handling in new QGC…?

taileron, 2020-06-11:
I am now out of the loop and have removed 10.15.xx from my computer because Apple suddenly blocks some of my old programs -> back in 10.13 and 10.12 using QGC 3.xx

dakejahl, 2020-08-01:
Is this still an issue?

taileron, 2020-08-01:
so for me no issue anymore I´m using QGC 3.4.4 (fullscreen 16:9 best rtsp I ever had) for monitoring and QGC 4.0.x for the rest, which still does bad rtsp. What´s about doing a special QGC release especially for flight monitoring with fullscreen support on all platforms, opacity control for menu and hud.

The1only, 2020-08-01:
I second that.

Best regard
Terje Nilsen
9Tek AS
Norway

DonLakeFlyer, 2020-08-07:
On daily builds double-click in video will go to fullscreen.

jaxxzer, 2020-08-27:
I've managed to reproduce the issues described here: increased latency and frame corruption on 4.0 compared to 3.5.

The issue seems to be dependent on the hardware: the vehicle hardware, network link hardware, and qgc host hardware. On some combinations 4.0 video streaming performs as well as 3.5, on other combinations, 3.5 performs well but 4.0 is awful.

I'm digging to narrow/find the problem.

BoKoppel, 2020-08-29:
@jaxxzer   Thanks for trying!!
I can test or log if that helps to narrow down more.
/ Bo

jaxxzer, 2020-09-16:
The issue appears to be that gstreamer 1.14.4 is not using hardware decoding on windows. Updating gstreamer to 1.18.0 resolves the video issues, and task manager shows the cpu load decreases significantly while the gpu load increases.

jaxxzer, 2020-09-16:
I can replicate the problem on gstreamer 1.18.0 also, if certain plugins are available. I think it is gstlibav.dll. The fix is probably to tell gstreamer the rank of decoder to use, prioritizing hardware decode if available. https://gstreamer.freedesktop.org/documentation/tutorials/playback/hardware-accelerated-video-decoding.html?gi-language=c

The1only, 2020-09-16:
Greate news... I hade the problems using Linux but I guess the same apply.!

Best regard
Terje Nilsen
9Tek AS
Norway

jaxxzer, 2020-09-18:
I do not have a laptop with intel or nvidia graphics. I only have tested with a ryzen laptop with radeon integrated graphics. 

On both 3.5 and 4.0 the pipeline is using avdec_h264 which is a software h264 decoder. 

There are multiple things at play. In most cases, the video on 4.0 is fine (although the latency is about 20% higher than 3.5). I am only able to see issues in a certain controlled test condition. In this particular setup, the video on 3.5 is fine, but the video on 4.0 is very poor with latency around 1-2s and very frequent video glitches/dropped frames/grey frames. The resource usage in task manager are the same between 3.5 and 4.0. This leads me to believe that there is some problem in the qgc streaming code between 3.5 and 4.0 - the qmlglsink being my primary suspect.

When I use gstreamer 1.18 with d3d11h264dec (available from 1.17) in place of avdec_h264, the issues in the 4.0 video are resolved. The cpu usage decreases by more than 50% and the gpu increases as expected. There might still be some problems in the qgc streaming code/video sink, but using the hardware decoder definitely removes the video glitches.

DonLakeFlyer, 2020-09-18:
> 1.18 with d3d11h264dec

What does this mean? Is this some special version of 1.18? Or are you just saying that 1.18 has a specific change in it which seems to fix the problem.

jaxxzer, 2020-09-21:
`d3d11h264dec` is the name for a hardware h264 decoder element for the pipeline. This element is available in gstreamer since 1.18. It at least removes the symptoms of the problem between 3.5 and 4.0 - both of which use software decoders, indicating that there is more going on. We should use hardware decoding wherever available in any case.

chrisfraser, 2020-09-29:
@jaxxzer I have a number of Windows, Mac and Linux computers around the office here. Please let me know if I can help test anything. I have a BlueROV setup and ready to go. I have a Teclast X6 Pro with Intel UHD Graphics 600. Video is pretty much unusable on 4.0, so that would be a good test.

My experience here is that I get get frame drop out using 4.0.X while recording on every computer I have tried. This includes the latest max-spec MacBook Pro 16" and a beefy Windows desktop. As a contrast I can watch the same video stream on a 5 year old, $100 generic Android tablet in VLC with no issues at all.

Happy to help in any way I can, but QT is not my game so with have to scale the learning curve. Will get a window dev environment setup over the weekend.

DonLakeFlyer, 2020-11-20:
@patrickelectric This is the hardware decoder problem right?

patrickelectric, 2020-11-20:
From what we have tested yes, the hardware decoder helps with the majority of these issues.

PR #9213 (2020-12-22): Enable and configure hardware decode priority 
- nvidia's hardwre code - tested by me
- vaapi hardware decode AMD - tested by @Williangalvani 

The older video implementation was using a specific pipeline, with the new video backend, now QGC is hostage of `decodebin` to take the decisions about the pipeline construction. 
Moving to `decodebin3` and enabling the hardware decode creates a much better scenario with a modern `decodebin` alternative that can handle a better construction and decision about the pipeline and elements that is used.  

Fix #8631
Needs #9099 to work under windows
closed 2020-12-22 by DonLakeFlyer