mavlink/qgroundcontrol #8632: symbol lookup error when launch QGroundcontrol.AppImage

matthewgan, 2020-04-03:
## Expected Behavior
Downloaded QGroundcontrol.AppImage and chmod, then it runs for a day.
After reboot and I installed a later version of gstreamer, I got this error when launching:

/tmp/.mount_EiEYlv/QGroundControl: symbol lookup error: /tmp/.mount_EiEYlv/QGroundControl: undefined symbol: gst_gl_display_wayland_new_with_display

## Current Behavior
QGroundControl refuse to launch

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.make install gstreamer1.0
2.launch QGC

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.04
- QGC Version: latest
- QGC build: stable
- Flight Controller: Pixhawk 2
- Autopilot (with version): Ardupilot Plane v4.0.5

I think QGC use gstreamer inside but how to fix this and let me relaunch QGC?

andrewvoznytsa, 2020-04-03:
Look on https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md

`We've standardized on GStreamer 1.14.4`

And perhaps rest of document will be also helpful. QGC is not trivial in terms on dependencies.

closed 2020-04-03 by andrewvoznytsa

DonLakeFlyer, 2020-04-03:
But this is using the prebuilt AppImage. There should be additional dependencies needed to install if you are using the AppImage should there? It should just work.

matthewgan, 2020-04-09:
I think the default gstreamer version with Ubuntu 18.04 stable is 1.14.5 and it works, I upgrade to 1.16.x, appImage refuse to launch and I downgrade back to 1.14.5 now, appImage still not working.

Understand the dependencies policies and I will try to rebuild from source.

dogmaphobic, 2020-04-09:
> I recall we were discussing that but I don't recall details. Can you please shed some light?

I don't remember either. GStreamer dependencies should always be included in the application "bundle". Linux has been the exception and I'm not sure why that's the case. That's why I didn't want to allow switching the build process to use pkg-config. It hardwires it to the version you have installed on the build machine.

```
LinuxBuild {
    QT += x11extras waylandclient
    CONFIG += link_pkgconfig
    packagesExist(gstreamer-1.0) {
        PKGCONFIG   += gstreamer-1.0  gstreamer-video-1.0 gstreamer-gl-1.0 egl
        CONFIG      += VideoEnabled
    }
```

Not sure if it makes sense to have an application bundle on Linux. There are way too many variations and incompatibilities in between versions of distributions. As is, if you want to download an Appimage, it's only supported on Ubuntu 18.04, which is our "standard". If you want to run on something else, you're better off building yourself from sources.

DonLakeFlyer, 2020-04-09:
> As is, if you want to download an Appimage, it's only supported on Ubuntu 18.04, which is our "standard".

I'm pretty sure it works on more distros than Ubuntu (or at least it used to) and the 18.04 requirement is a new thing. The only thing I know about this is hearing from people about using it and complaints about things not work. I don't know enough about Linux to know what's truly correct. Given that, to me it seems like the gstreamer we depend on should be bundled in the appimage.

Tech2morrow, 2022-07-29:
I am facing this issue for longgg

Tech2morrow, 2022-07-29:
> Look on https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoReceiver/README.md
> 
> `We've standardized on GStreamer 1.14.4`
> 
> And perhaps rest of document will be also helpful. QGC is not trivial in terms on dependencies.

anusha-bylla, 2024-12-20:
“Hi, I am new to Qt Creator. I successfully ran QGroundControl on Windows, but now I am trying to run QGroundControl for Android in Qt Creator, and I am encountering the following error:”
can you please help me to solve this error…
[image]
[image]
[image]