mavlink/qgroundcontrol #8635: Android Version - Offline Map Import - No files

ve3otk, 2020-04-05:
## Expected Behavior
Offline Maps tab - select Import Maps, Append to existing tiles, should be presented with a file browser to browse to SD card to import tile sets

## Current Behavior
No means of browsing to where tile sets are stored

## Steps to Reproduce:
As above
[image]

## System Information

DJI Crystal Sky CS550, Android 5.1.1
QCG 4.0.5

DonLakeFlyer, 2020-04-05:
Mobile QGC never presents a file picker UI. It only shows file from the standard locations which in this case is internal storage (not SD) QGroundControl/Missions. 

This is because the android version does not have access to a standard OS level file picker ui. I'll take a look at this again to see if it can be made better but until then you need to move the files manually to the known location

DonLakeFlyer, 2020-04-05:
> I'll take a look at this again to see if it can be made better but until then you need to move the files manually to the known location

Took another look. The file picker available is pretty much unusable still.

DonLakeFlyer, 2020-04-05:
So as it currently stands there is no good way to make this better

closed 2020-04-05 by DonLakeFlyer

ve3otk, 2020-04-05:
Might be worth putting the path for Importing tile in the online documentation - if it was there, was unable to locate it earlier today.

Would it be possible to enter an alternate import path on the General tab in the settings?

Anyway, once you mentioned the location the app is looking for, was able to import the map tiles, then re-connect to the computer and delete the map tile files to free up internal storage. Thanks!

DonLakeFlyer, 2020-04-05:
> Would it be possible to enter an alternate import path on the General tab in the settings?

Doesn't seem to make sense to do that for a single case. Manually moving files from an SD card to the specific locations is pretty much what you need to do with everything. Same thing to get a Plan you created on desktop to your tablet.

> Might be worth putting the path for Importing tile in the online documentation

If you look at my linked pull you'll see that path is show now. I'm going to look at docs as well.

DonLakeFlyer, 2020-04-06:
@hamishwillee So I want to write something up about managing file movement on mobile builds. But I'm not really sure where to link it in to. Any ideas?

hamishwillee, 2020-04-07:
Hi @DonLakeFlyer There is nowhere good. IMO it can really only be under the Support section - as a kind of non-standard out of band support. 
Possibly in https://docs.qgroundcontrol.com/en/Support/troubleshooting_qgc.html . 

Alternatively a new peer topic "Working with QGC Mobile". Would kind of be handy because then you could list some of the other things that aren't supported that you're always being asked about.