mavlink/qgroundcontrol #8639: Land start marker displayed on previous waypoint compared to what is executed

DanielePettenuzzo, 2020-04-06:
## Expected Behavior
If I set the land start marker to a specific waypoint in the plan UI I expect the vehicle to go to this waypoint to start a landing procedure (eg. if I trigger and RTL during the mission).

## Current Behavior
The vehicle starts the landing maneuver on the next waypoint compared to what was displayed in the mission plan UI. 

## Steps to Reproduce:
1. Plan a VTOL mission that has a land start marker
2. Trigger an RTL before the mission in over
3. The vehicle will not go to the waypoint that had the land start associated to it but to the next one instead

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. Arch Linux]
- QGC Version: master from April 6, 2020 (git hash: 793d8d9632694d64153c8)
- QGC build: self-built from source
- Flight Controller: standard vtol in gazebo
- Autopilot (with version): master from April 6, 2020 (git hash: 92de0048410cf07c2c8)

## Log Files and Screenshots
Mission planning:
[image]
Mission execution:
[image]

DanielePettenuzzo, 2020-04-06:
FYI @RomanBapst @sfuhrer

DonLakeFlyer, 2020-04-06:
Hmm, that's a tricky one. For mission commands which do not include a position the visual representation is to highlight the previous item which has a location. For these types of commands this indicates that those commands will be executed "after" that waypoint. You can have commands after a DO_LAND_START such as bringing the gimbal back to neutral and stopping camera commands. All of which would happen prior to the new WP command. If you highlighted instead the "next" waypoint that would be even less correct since it would seem to show that the commands are the DO_LAND_START are not executed.

So yes it's odd for this specific case, but it's more correct from a consistent standpoint of how QGC deals with this.

DonLakeFlyer, 2020-04-06:
> For mission commands which do not include a position the visual representation is to highlight the previous item which has a location.

Would love to here opinions on whether in general that makes sense.

> So yes it's odd for this specific case, but it's more correct from a consistent standpoint of how QGC deals with this.

After playing around with this some. I don't think the consistency argument here makes sense. Nobody will understand that. Also the problems with having additional commands between DO_LAND_START and the next WP may or may not be that common. So I think it makes sense to one-off DO_LAND_START to change the selection mechanism. Will be a  bit tricky since it could lead to no selection which current code can't handle but I can figure that out.

Antiheavy, 2020-04-06:
@DanielePettenuzzo how would you feel if QGC had a built-in "VTOL Landing Pattern" similar to the current Fixed Wing Landing Pattern?  In this case the DO_Land_Start commend is hidden away and the user probably doesn't even realize it's there, all they know is the vehicle always goes to the start of the landing pattern when commanded to Return mode (assuming RTL_TYPE=1).

I suspect if a VTOL Landing Pattern were built-in it would become very rare for VTOL users to manually plan out a landing patterns and ever need to use the Do_Land_Start command.  Maybe only advanced users would do so.

DonLakeFlyer, 2020-04-07:
> I suspect if a VTOL Landing Pattern were built-in it would become very rare for VTOL users to manually plan out a landing patterns and ever need to use the Do_Land_Start command.

I talked with @sanderux about this exact thing a few weeks back. It didn't seem like we could come to a concensus on a standard pattern that would work for most cases. That said I would love to create a VTOL Landing Pattern if someone can explain to me what it should look like to fit a decent set of use cases.

RomanBapst, 2020-04-07:
@DonLakeFlyer I pretty much agree with what @Antiheavy said. It would be super cool if there is a landing pattern as an item which you can simply select, without having to manually add land start marker and all that stuff.
I'm open to other suggestions but here is what we have used so far and found helpful:

**Loiter to altitude and exit loiter on a tangent to landing waypoint**
Currently we achieve this behavior by adding 3 waypoints manually:
1) Land start marker
2) Loiter to altitude (check the "exit on tangent box")
3) Land wayoint

This approach has the following advantages:
- during the loiter wind becomes observable which is nice to know
- you can potentially decrease your airspeed during the loiter
- you exit the loiter in a straight line towards the land waypoint which gives you a clean entry to the transition

**Additional Notes**
As of recent PX4 controls deceleration and path during the backtransition. This means that the ground station can actually limit the minimum distance between the landing waypoint and the loiter.
The target deceleration during the transition is **VT_B_DEC_MSS** and together with FW_AIRSPD_TRIM and known wind conditions you could calculate the distance the vehicle needs to transition to hover.

Of course we could always extend the functionality and offer more than one type of landing patterns.

DanielePettenuzzo, 2020-04-08:
@RomanBapst I think your proposal makes sense

DonLakeFlyer, 2020-04-08:
OK, I'll add a VTOL landing pattern and we can continue discussion after people play around with it.

DonLakeFlyer, 2020-04-15:
Can anyone here with a VTOL test this? I've done as much SITL testing as I can.

sfuhrer, 2020-04-16:
> Can anyone here with a VTOL test this? I've done as much SITL testing as I can.

I've given some feedback in the PR. Lets bring it to AGS asap!

DonLakeFlyer, 2020-11-06:
VTOL landing pattern has been in for ages

closed 2020-11-06 by DonLakeFlyer