mavlink/qgroundcontrol #8640: Add Bluetooth Comm Link Connection Type to Desktop QGC

Arktix13, 2020-04-07:
Trying to connect Team Black Sheep Crossfire Tx (full size module) to Windows 10 laptop via bluetooth. Looks like bluetooth comm link is not supported on any version of desktop, only on Android. Requesting bluetooth connection be added to comm link options for desktop QGC.

- I'm trying to use this Crossfire bluetooth connection with a large multirotor running a custom version of PX4 (FreeFly Alta X).

- It looks like this works on the Android version of QGC, but no other version (tried iOS and Windows 10).

- I need the bluetooth connection to send and receive MAVLink messages between the laptop QGC and aircraft.

Note: When I got this working with the Android version of QGC it required setting the MAVLink version on the Alta X to MAVLink v1. Apparently TBS Crossfire does not currently support MAVLink v2.

bys1123, 2020-04-07:
Bluetooth on Desktop is also a COM port
[image]

Arktix13, 2020-04-07:
I tried selecting different COM ports, but none of them worked. Have you ever connected a Crossfire Tx to desktop QGC?

tenten8401, 2020-04-12:
I can connect via bluetooth to desktop QGC under linux using /dev/rfcomm0 and `rfcomm connect rfcomm0 MACADDRESS`

mamariomiamo, 2021-07-19:
> I can connect via bluetooth to desktop QGC under linux using /dev/rfcomm0 and `rfcomm connect rfcomm0 MACADDRESS`

Hello, are you PX4 firmware?

karthikdesai, 2021-07-21:
The trick with the `sudo rfcomm connect rfcomm0 MACADDRESS` seams to half work, once you do it a couple of times.

But the connection does not get established. What are the baudrate and stop bits?
We are using the latest TBS Firmware available running with ArduCopter 4.0.7.

closed 2024-06-04 by HTRamsey