mavlink/qgroundcontrol #8646: Qgroundcontrol "This app cannot run on this PC"

pbecchi, 2020-04-09:
I have installed QGroundControl with Windows 10 32bit on a laptop ASUS T200.
Installation is OK, but the application dont run, i get the message :
"This app cannot run on this PC, contact app publisher."

I have tryed  all suggested method to overcome this issue with no success

dogmaphobic, 2020-04-09:
QGC is a 64-bit application. It will not run on a 32-bit version of Windows.

closed 2020-04-09 by dogmaphobic

pbecchi, 2020-04-10:
Thanks.Good to know : i coudn't find this anywhere in the documentation!

hamishwillee, 2020-04-14:
@pbecchi Sorry about that. Don added a note about this in the installation instructions following your post: https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html#windows