mavlink/qgroundcontrol #8659: QGC wont connect on UDP on Android phone

Hwurzburg, 2020-04-15:
using both stable and daily, will connect to any PC over WIFI using ESP-01 flashed with ArduPilot firmware for ESP8266 ....connects to both MP and QGC in Win10, and QGC in Ubuntu18.04.....but will not on my Motorola Z2 Play phone running Android 8.0.0 and QGC...QGC on phone works fine with BT connections....

will occasionally connect, but only 1 out of 20-30....tried manual UDP connection also...

console shows many tile fetch attempts, but there is no internet...

DonLakeFlyer, 2020-04-22:
> will occasionally connect, but only 1 out of 20-30

This is the part that is really odd. I wouldn't expect partial failure. I'll need to make you a build with some additional logging. Need to see if anything is coming in on the UDP port.

Hwurzburg, 2020-04-22:
Maybe something else is interfering.....I tried in on Goggle Pixel3+ running Android 9 which is dedicated to the Mavic Mini travel pack and doesnt have internet connectivity, just wifi, and it performs flawlessly...and maybe its a Motorola Phone issue somehow, because my previous Razor had the same issue 2 years ago...in fact I switched my DL TX from an ESP8266 back to BT because I couldn't realiably connect to QGC

Hwurzburg, 2020-04-22:
If I am the only one complaining about it, just close this, but I really would like to use my normal cellphone instead of carrying a dedicated one for ground station use...

DonLakeFlyer, 2020-04-22:
Can you try again and check the log for a line starting with "Adding target". This should get spit out when the UDP port gets the first incoming traffic and tells you the IP/Port it came from. It then uses this to talk back to the vehicle though. If you don't ever get that then it means there is no data coming in on port 14550.

Hwurzburg, 2020-04-22:
[image]
This am I tried and could not get it to fail...return Port is 14555 as expected...but then I powered things down...came back a few minutes later and phone would not connect...my PC's could,my Pixel3 could...above is the console output

RhinoAus, 2020-08-25:
I was also having this issue on my Pixel 3. I found the issue was due to the 14550 port being occupied by some other app. I have no idea which one, but changing from 14550 to 12345 fixed my problem.

Is there a way to add some sort of reporting if the UDP port is already taken?

closed 2024-06-02 by HTRamsey