mavlink/qgroundcontrol #8671: Motor test stops after 3 seconds

jancoow, 2020-04-18:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
The motor should just keep spinning after I set a manual throttle value for testing. 

## Current Behavior
Please describe what happens instead of the expected behavior.
After around 3 seconds all motors stop spinning and the slider is reset to 0

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Go to motor setup
2. Mark the checkbox that you removed the propellors
3. Set a motor to a value with the slider
4. Wait a few seconds and it will be turned off

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2020-04-18:
This is working as designed. It turns off automatically for safety reasons.

jancoow, 2020-04-18:
What safety, if you already checked the box that you take of your props? Can I turn of this safety? Or can I at least increase the time?

DonLakeFlyer, 2020-04-20:
I'd have to find the original issues/pulls associated with this. There much discussion around the safety of motor test. What you see now is the end result fo those dsicussions.

DonLakeFlyer, 2020-04-20:
The main focus for motor test is to verify correct motor ordering. Hence you just need to see it spin up to verify.

jancoow, 2020-04-21:
Ah right, I got it. Might be worth putting a image next to it with motor order and motor numbering. 

The problem is that I want to test the vibrations generated by each single motor, but I'm unable to find a way to turn one motor on for longer then 3 seconds.

DonLakeFlyer, 2020-04-21:
> The problem is that I want to test the vibrations generated by each single motor, but I'm unable to find a way to turn one motor on for longer then 3 seconds.

I think the motor test command available from mavlink console allow more options to do things line that.

> Might be worth putting a image next to it with motor order and motor numbering.

Yeah it was there. And in fact the code for it is already there as well but it was turned off. Because it was pointed out that the airframe config can create non-standard motor ordering. I'll think about looking at that again since that likely happens almost never.

DonLakeFlyer, 2020-04-21:
I'm keeping this issue open and will use it to tack motor number visuals.

jancoow, 2020-04-21:
Cool, thanks! You might wanna add some description to it as well, explaining it will turn off after 3 seconds for safety reasons. That might make it more clear :)

closed 2026-10-05 by github-actions[bot]