mavlink/qgroundcontrol #8687: QGroundControl does not saving video stream to file on android device

dezertefer, 2020-04-23:
Hello there, I was working with QGroundControl some time and found some bug. Gstreamer works just fine on Windows machine and Ubuntu, however It doesn't save the video in file on Android machine.
Tried Android versions 6.0 and 9.0
Getting "Failed to link valve and file sink" message.

DonLakeFlyer, 2020-04-23:
Is this stable or daily?

dezertefer, 2020-04-23:
Used master and release which is on Google play. Hope it contains stable version if not I may check it and return here with the result.

DonLakeFlyer, 2020-04-23:
Ths stable on Google Play is slightly behind. Since your seeing it on master as well. That would indicate the problem is on both places which means a fix needs to go into Stable as well.

dezertefer, 2020-04-23:
Okay, I was playing around with Gstreamer. Everything seems to be implemented correctly (according to some web search), but doesn't work. So I didn't fix that. Hope somebody will figure that out.

andrewvoznytsa, 2020-04-25:
@dezertefer Stable_V4.0 and master have potential fixes for your issue. If you can't build yourself then just wait for new daily build and let us know if issues was fixed.

dezertefer, 2020-04-25:
> @dezertefer Stable_V4.0 and master have potential fixes for your issue. If you can't build yourself then just wait for new daily build and let us know if issue was fixed.

Thank you for the tip, going to try.

dezertefer, 2020-04-25:
> @dezertefer Stable_V4.0 and master have potential fixes for your issue. If you can't build yourself then just wait for new daily build and let us know if issue was fixed.

It works. Showed some errors according to save path. But after the remove of the previous video folder and rebuild it started to save files. Thank you. May I ask you something? I saw the issue with delays and gray frames for the stream on android and IOS devices, I suppose problem there is much more complex right ?

andrewvoznytsa, 2020-04-26:
Closing due to confirmed fix.

closed 2020-04-26 by andrewvoznytsa

andrewvoznytsa, 2020-04-26:
In my opinion delays and grey frames are not QGC's issues - it is side effect of some strange gstreamer decisions (internal algorithms). They definitely can work in other way but they work as they work and we have what we have.

There are ways to workaround these issues but their complexity is starting around a few days/week of development. I think it would be easier to try gstreamer 1.16 (or even latest) to see if there are potential gains/internal fixes/improvements which affects delay and/or grey frames. We need update to 1.16 due to other issues which are promised to be fixed there. So it is worth to try. And we'd appreciate any help there.

dezertefer, 2020-04-26:
I already tried 1.16  3-4 days ago, I don’t remember If I used latest version with all your fixes, but it didn’t fix delays. So I need to check again with latest release to be exact.