mavlink/qgroundcontrol #8688: mavlink inspector SERVO_OUTPUT_RAW display error

vivienJ, 2020-04-23:
## Current Behavior
When the UAV has main and aux mixers it send two ORB actuator_outputs. These ORBS are displayed on the same plot on MAVLINK inspector -> SERVO_OUTPUT_RAW.

For exemple, on the screenshot, the main output 1 = 900 and the aux output 1 = 1500, QGC tries to display them both at the same time.

## System Information
- Operating System: Ubuntu 18
- QGC Version: 4.0.6
- QGC : build:stable
- Flight Controller: Pixhawk3Pro and Pixhawk4
- Autopilot (with version):  PX4 1.10.1

## Detailed Description
[image]

DonLakeFlyer, 2020-04-24:
It's just not possible to graph telemetry that works like that. And supporting it would be quite complex. So unlikely to happen anytime soon. You may need to go to vehicle logs to get that level of detail.

closed 2020-04-24 by DonLakeFlyer

JacobCrabill, 2020-10-05:
@DonLakeFlyer the previous QGC release (3.x) could do this (plot values from port "0" vs. port "1" independently); what changed?  Can we bring that back?

DonLakeFlyer, 2020-10-05:
> what changed?

The old graphing code couldn't run in the new version of QGC. So it was completely rewritten from scratch. Along the way this feature was lost. At some point I'll try to get it back but it's unclear how much work it will take.

KnightHawk06, 2024-06-11:
Is this still an issue in QGC v4.3.0 and PX4 1.14?

ROSStargh, 2024-06-28:
> Is this still an issue in QGC v4.3.0 and PX4 1.14?

Confirmed still an issue.