mavlink/qgroundcontrol #8691: [Feature request (or bug fix?)] Force Landing Pattern to be at the end of the mission

Antiheavy, 2020-04-25:
We've had a number of users get bitten by this recently.  At the moment it is possible to "insert" a landing pattern in the middle of a mission.  This tends to have all sort of bad effects when the user tries to fly this mission.

[image]

QGC already only allows 1 landing pattern and it prevents you from adding waypoints after a Landing Pattern.  However, it doesn't prevent you from inserting a landing pattern in the middle of a mission.  

It is nice to be able to add a Landing Pattern at any time.  Can we force the one and only Landing Pattern to always be at the end, no matter when or how it was added?

DonLakeFlyer, 2020-04-25:
What Todd describes is exactly how it's "supposed" to work. Bug city.

Antiheavy, 2020-05-06:
to repo:

1) start a Blank mission plan
2) plan takeoff
3) plan a survey or any number of waypoints
4) click on the takeoff or some waypoing/survey on the right-hand list (one that is NOT at the end of the mission)
5) plan a landing pattern, notice it is not at the end.

DonLakeFlyer, 2020-05-06:
You have to add a Pattern to get it to screw up

closed 2020-11-06 by DonLakeFlyer