mavlink/qgroundcontrol #8694: Grid survey always looking at the horizon (has

remster, 2020-04-28:
## Expected Behavior
Grid (lawnmower) survey should at least point the gimbal straight down and ideally it should let the client/user set it.

## Current Behavior
Drone snaps photos of the horizon when traversing the survey.

## Detailed Description
There is this bit of code in the `MissionController`
```
        if (_controllerVehicle->firmwarePlugin()->hasGimbal(_controllerVehicle, rollSupported, pitchSupported, yawSupported) && pitchSupported) {
            // If the user already specified a gimbal angle leave it alone
            if (!cameraSection->specifyGimbal()) {
                cameraSection->setSpecifyGimbal(true);
                cameraSection->gimbalPitch()->setRawValue(-90.0);
            }
        }
```
that sets the gimbal to look "straight down", but only under some conditions. One of these conditions is: `_controllerVehicle->firmwarePlugin()->hasGimbal`
That:
- never returns true - there is just [one implementation of hasGimbal](https://github.com/mavlink/qgroundcontrol/blob/09a98e34a0aa3e182c6cb2b2bac861d3cf4a7659/src/FirmwarePlugin/FirmwarePlugin.cc#L775) in QGC and no FirmwarePlugin overrides it.
- Suppose the vehicle has no gimbal. What's the point of using it for the grid survey at all? What's the point of checking that it does? Suggest to remove the check altogether.

DonLakeFlyer, 2020-04-28:
Using mavlink there is no way to determine if a vehicle has a gimbal, hence standard QGC always assume no gimbal. Custom builds can create there own firmware plugin overrides to tell QGC that their specific vehicle has a gimbal which will in turn make this work. A vehicle does not need a gimbal to do surveys. Most fixed wing aircraft with cameras do not have gimbals. The camera is fixed point straight down. Just like there is no way to know through mavlink if a vehicle has a gimbal, there is also no way to know if the vehicle actually has a camera and also no way to know if that camera if it is there what orientation it is in by default. So for regular QGC users have to manually deal with the gimbal as needed.

closed 2020-04-28 by DonLakeFlyer

hamishwillee, 2020-04-28:
@DonLakeFlyer The [Gimbal API V2](https://mavlink.io/en/services/gimbal_v2.html) does allow you to know if there is a gimbal present - see [discovery](https://mavlink.io/en/services/gimbal_v2.html#discovery). This is WIP of course but is being implemented in MAVSDK as we speak. Julian Oes is person who knows most about progress of this in PX4 firmware.