mavlink/qgroundcontrol #8696: Using msvc2017 and qt5.12 will report c2220 errors.

Phillweston, 2020-04-28:
Using msvc2017 and qt5.12 will report c2220 errors.

Error: c2220: warning treated as error - no 'object' file generated

The source code cannot be compiled due to this error

dezertefer, 2020-05-09:
Got the same issue. Latest code for today and Qt 5.12.8

dezertefer, 2020-05-09:
UPD: works just fine on Linux, error occurs on Windows machine

dezertefer, 2020-05-09:
[image]

CONFIG += warn_off

this line fixed issue for me in .pro file

DonLakeFlyer, 2020-05-13:
FYI: This is happening because it looks like you don't have gstreamer installed. I'm working to fix it up. That said did you really want to build without video support?

DonLakeFlyer, 2020-05-13:
Also QGS should be built with 5.12.6 which is what the CI released build uses. Make sure to follow the dev guide instructions exactly or you may run into problems.

dezertefer, 2020-05-13:
> Also QGS should be built with 5.12.6 which is what the CI released build uses. Make sure to follow the dev guide instructions exactly or you may run into problems.

I have two packages 5.12.6 and 5.12.8 worked well several weeks ago. Videostreamer doesn't change it, errors start to appear from SimpleMissionItem with videostreamer installed and in many other classes.

DonLakeFlyer, 2020-05-13:
> Videostreamer doesn't change it, errors start to appear from SimpleMissionItem with videostreamer installed and in many other classes.

You'll still get the errors but it won't fail the build. The SimpleMissionItem thing is the VS compiler being stupid. VS seems to be dumb to a case statement which uses all the possible values which means there is no possibility of of not returning a value.

Using a version of Qt other than the supported version "might" work. But only might. You can run into subtle bugs which only show up in that version. Happens all the time. You loose the benefit of using a tested configuration which is backed up by 1000's of active users reporting problems.

dezertefer, 2020-05-13:
> You'll still get the errors but it won't fail the build. The SimpleMissionItem thing is the VS compiler being stupid. VS seems to be dumb to a case statement which uses all the possible values which [...]

The thing is that it did, it failed the build with GStreamer and right Qt setup I think this is some kind of VS compiler issue, it messes QObjects, creates warnings according to them and than recognizes these warnings as errors and stops the build. I will try to recreate this behaviour on second windows machine If it may help you.

DonLakeFlyer, 2020-05-13:
> The thing is that it did, it failed the build with GStreamer and right Qt setup

I can't reproduce that. In fact I discovered that the Video part of the build was turning off all warning as errors no matter what you set it in other places, which is completely wrong.

DonLakeFlyer, 2020-05-13:
The windows fix is here: https://github.com/DonLakeFlyer/qgroundcontrol/tree/CompilerWarningsBackOn. But I"m still testing the changes on other OS since I removed the hack in Video which was turning everything off it likely is popping errors on other OS as well. Need to test that out before submitting a pull.

DonLakeFlyer, 2020-05-15:
Master should finally be clean now

closed 2020-05-15 by DonLakeFlyer