mavlink/qgroundcontrol #8698: Feedback: Plan - Terrain Profile/Collision support

DonLakeFlyer, 2020-04-28:
The Terrain Profile Chart is an update to the view at the bottom of the Plan View accessed by clicking the "T" button if you have it hidden. The current state of this new feature is that it will take some time to wring it out. Adding support for it was very complex and require much code change. Hence it will likely be "quirky" for a while. I already have a big long list of fixes/changes/features which i still need to work on. 

For now I would like to collect feedback on how it works. Later I will be asking for help on finding bugs in it once I've had more time to find fix/bugs on my own. So until then please, don't enter a pile of issues on bugs on it. I know it's buggy. That said if it crashes or something goes horribly wrong then report it. I would describe it as "quirky" at this point. This is only a visual display so the actual generation/upload of Plans should not be affected.

Features which I already know are missing:
* Ability to zoom in/out and scroll
* Correct visualization of Takeoff and RTL flight path segments
* Better indication of when all terrain information has become available and the Terrain Profile has reached it final form
* Ability to prevent a save/upload if there is a terrain collision

# Survey Terrain Follow Example
[image]

## Map Mission Elements
* The display af the mission over the map is essentially the same as before except for the "red" coloring of items which are colliding with terrain.
* Patterns for which any portion of their flight path collides with terrain will change there color from green to  red. In the example you can see a Survey for which two of the transects collide with terrain. Because of this the entire survey area is shown in red.
* Simple flight paths segments from one waypoint to another will also change from orange to red if there is a terrain collision. You can see that in the flight path segment from WP 127 at survey exit to WP 128.

## Terrain Profile Chart
This is the chart you see at the bottom of the window. The "T" button will show/hide it. It updates dynamically as you manipulate the mission.

### Chart Axes
* Y Axis - Runs from the minimum to maximum AMSL altitude within the mission. This includes both the flight path heights as well as terrain heights. This means the top portion of the chart will always be either the highest flight path segment or highest terrain point.
* X Axis - Runs from 0 to the total distance of the mission from takeoff the landing.
* It is important to note that all heights in the charts are AMSL.

### Line Series Information
There are multiple sets of lines shown in the chart with each color showing a specific piece of information:
* Green lines: This is the Terrain Profile throughout the Plan
* Orange lines: This is the Flight Path Profile the vehicle will take
* Red lines: This is a flight path segment which is colliding with terrain
* Yellow lines: This is a portion of the terrain profile for which the terrain data is still loading. It will always be at the bottom of the display.

#### Terrain Profile
This shows you the profile of the terrain under your flight path for the entire mission. Terrain heights are queried from the QGC terrain cache and may need to be pulled from the internet if QGC doesn't already have the needed information cached. Because of this the terrain profile may take a few seconds to fully display. While there a missing bits of terrain information you will see yellow line segment at the bottom of the chart which indicates this portion of the terrain profile is still missing data.

There are also cases where a flight path segment requires terrain data in order to display.  This can happen if you are using "Above Terrain" or "Terrain Frame". Until the terrain data comes back the AMSL heights for those items is unknown. For these you should see the yellow missing terrain line and there will missing flight path segments in that portion of the mission.

#### Flight Path
The flight path shows you the altitude profile of the entire mission. If any of the flight segments collide with terrain that flight segment will be show in red. This is true for simple flight segments like flying from one added waypiont to another. Or the interior of complex patterns such as Surveys, Corridor Scans and so forth. In the example above you can see that two of the transects collide with terrain. All the way to the right you can also see flight segment 127-128 defined by simple waypoints also collides with terrain.

#### Mission Item Indicators
You can also see individual mission items in the chart. They will be placed at the distance they occur with the mission. You can click on them (like before) to select that item. The currently select item is shown in green (also like before).

Patterns (Survey, Corridor Scan, ...) are also shown as a green rectangle and the pattern name at the bottom of the display. The position and width of the rectangle corresponds to the distance based position of the pattern within the entire mission. You can click on this indicator to make the pattern the current item.

DonLakeFlyer, 2020-04-28:
I'll update this comment as new things are determined which should be added/changed.

Missing Features:
* Ability to zoom in/out and scroll
* Correct visualization of Takeoff and RTL flight path segments
* Better indication of when all terrain information has become available and the Terrain Profile has reached it final form
* Ability to prevent a save/upload if there is a terrain collision

hamishwillee, 2020-04-28:
1. How do you map the problem region between terrain and map?
2. How do you fix it? Do you have to increase altitude of the whole pattern, or can you do this in parts/segments?

I'd really like to be able to see problematic terrain in the map view itself along the path. Ideally following same convention as the terrain view. Perhaps even some beacons showing problem areas during planning based on your current default mission item/pattern height.

Antiheavy, 2020-04-29:
Looking pretty awesome.

fingadar, 2020-04-29:
Within this changes, will it be possible to use an onboard rangefinder (i.e. a lidar) with ardupilot in surveys?

DonLakeFlyer, 2020-04-29:
> Within this changes, will it be possible to use an onboard rangefinder (i.e. a lidar) with ardupilot in surveys?

That is unrelated to this change. The discussion around that is more involved and relates to where QGC support MAV_FRAME=MAV_FRAME_GLOBAL_TERRAIN_ALT and where QGC supports generating terrain follow itself. Can you enter a separate issue for that and we can continue there?

DonLakeFlyer, 2020-04-29:
> 1. How do you map the problem region between terrain and map?

You can click the item in the terrain profile and it will select the item in the map.

> 2\. How do you fix it? Do you have to increase altitude of the whole pattern, or can you do this in parts/segments?

You either need to increase the entire altitude of the survey giving up ground resolution. Or for terrain follow you need to adjust your tolerances to be lower. Or both. For terrain follow it is possible to run into cases where it is not possible to create the mission items which match you tolerances and not hit terrain. This can happen if you are trying to fly too close to terrain and there is no way to make that possible if the terrain under the flight path increases/decreased too quickly.

> 2\. or can you do this in parts/segments?

That would not be a survey as defined by the survey, corridor scan features. A survey is used to fly at a set altitude above ground. All transects of the survey must be at the same altitude above ground. This allows you to take images at a specific ground resolution which you can stitch together into a single geo-referenced image. If one of the transects is at a different height none of that works.

valmorgan2, 2024-05-28:
> I'll update this comment as new things are determined which should be added/changed.
>
> Missing Features:
>
> * Ability to zoom in/out and scroll
> * Correct visualization of Takeoff and RTL flight path [...]

@DonLakeFlyer Was the "> * Correct visualization of Takeoff and RTL flight path segments" ever solved? 

Thank you!

HangarZer0, 2024-09-20:
> @DonLakeFlyer Was the "> * Correct visualization of Takeoff and RTL flight path segments" ever solved?
>
> Thank you!

@DonLakeFlyer Following up on the takeoff and RTL path segments in terrain follow mode. Currently the terrain graph in the mission planning interface seems to show those segments following a relative height and impacting the ground during RTL from a terrain following survey, even though RTL_ALT_TYPE has "Terrain" selected. I want to test out the RTL with both WPNAV_RFND_USE enabled and disabled at my field site but I'm nervous about it since the terrain graph shows an impact during RTL with a relative height flight path indicated for the RTL segment and also the Takeoff/ferry segment. Also, when double checking with a .KML export to Google Earth, the RTL segment does not appear to be included in order to analyze whether an impact is actually indicated or not during takeoff and RTL. The takeoff segment appears in the .KML but it also looks like a relative height flight segment versus a terrain-followed segment. Any help is appreciated. Thanks!

closed 2026-10-05 by github-actions[bot]