mavlink/qgroundcontrol #8704: Load Kml: Displacement error

Runge123, 2020-04-29:
## Expected Behavior
Load kml add correct position

## Current Behavior
When I load one kml in QGC, it looks displaced

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. I want to load kml (crs 4326) in Qgroundcontrol. Files in https://www.dropbox.com/sh/woar0i24qjzmjjq/AABgmsNVQvHTqOY12aeKkCyZa?dl=0
2. In Qgroundcontrol I go to pattern > survey > load kml/shp
3. The polygon (kml) loaded in Qgroundcontrol is shifted

## System Information
When posting bug reports, include the following information
- Operating System: windows 10
- QGC Version: 4.0.6

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
All files in dropbox link

DonLakeFlyer, 2020-04-30:
Hmm it seems like a possible problem with the Bing (QGC default) map tiles being georeferenced incorrectly. If you switch the map provider to Google you'll see the polygon now shows up in the right place.

@dogmaphobic Is there a possibility this is a QGC problem. Or is it just bad tiles?

Runge123, 2020-04-30:
I confirm that the above kml, as well as the following output kml created with QGC from it (https://www.dropbox.com/s/wa3imdfiufd1f9z/BertAOIpratDronica_epsg4326_v2.kml?dl=0), are correctly positioned in QGC if Google is selected as map provider. Also, the output kml is correctley positioned in GE and QGIS

dogmaphobic, 2020-04-30:
Coordinate system mismatch? When you load it in QGIS, can you tell what is the coordinate system used? (it should also be within the kml file). I have no idea if Qt even has the notion of coordinate systems. I don't have the time now to look. I need to see the kml file and see how it is defined. I will do that when I have a chance.

DonLakeFlyer, 2020-04-30:
Nope. The coordinates in the KML are all good. If you load it in Google Earth you can see where it goes. You can even load it in QGC, and then Save to KML and the load that back in Google Earth and see it goes back to the correct place. It's only screwed up if you are using Bing Maps. You can just switch map providers, go back to plan and see it move.

DonLakeFlyer, 2020-04-30:
This problem has nothing to do with KML. Do this:
* Boot QGC
* Goto Plan
* Add a takeoff item next to some landmark. I used my own house.
* Change between Google/Bing. Note that the takeoff item stays at the same place
* Now change the takeoff item lat/lon to this:
[image]
* Note the position visually change map provider look at again
* I get these differences. Can't remember which one is google or bing
[image]
[image]

I think it just might be map provider georeference errors as opposed to anything QGC could b getting wrong.

closed 2020-11-06 by DonLakeFlyer