mavlink/qgroundcontrol #8707: Terrain Following in Survey Mission

chris89116, 2020-04-30:
Is it possible to fly a survey mission while using an on-board laser rangefinder to follow terrain ?
My mission requires flight at low altitude with tight altitude compliance over rolling terrain. I have experimented using the DEM terrain height but it is just not accurate enough, even using 1 meter DEM data.

[https://discuss.px4.io/t/terrain-following-in-survey-mission/11681](url)

Any additional context you can provide will make the feature easier to evaluate (e.g. mockups, detailed specification, etc.)

DonLakeFlyer, 2020-04-30:
> Is it possible to fly a survey mission while using an on-board laser rangefinder to follow terrain ?

The current answer is no. Initially that was because QGC did not support mavlinks terrain frame for mission commands. The current daily builds do support that now that including uploading terrain tiles to the vehicle if requested (for non-lidar use cases).

When I made those changes I had discussions with folks using terrain follow heavily. The result of those discussions was that the gcs side generated terrain follow that QGC has for their use cases was better than the mavlink terrain frame. The reason being it allows them to visual the actual flight path of the vehicle with respect to maximum climb/descent rates of a specific vehicle. That discussion led me to think that supporting terrain frame within "patterns" was not needed since the existing support was actually better/safer.

The thing those discussions did not cover was your specific use case. Which is also valid and useful and invalidates the thinking that support terrain frame within patterns should not be included.

So anyway, just wanted to go through some history so anyone else looking at this knows what is going on.

All of that said, since QGC does now support terrain frame internally, it is possible (whereas previously it was not) to add the feature you want. I'll hopefully have some time to get to it soon and hopefully you can help test it out.

chris89116, 2020-04-30:
Yes it would be with pleasure.

DonLakeFlyer, 2020-04-30:
First step to verify would be that you can fly simple terrain frame missions without patterns using lidar. Change the altitude mode on the items to Terrain Frame. That is possible in current daily. I tested all of this using terrain data being sent to the vehicle. But it hasn't been tested with ArduPilot lidar terrain frame support.

chris89116, 2020-04-30:
I am ready to test on the other hand my english is not very good and I am not yet an expert in the field I prefer to prevent!
When you say : You can fly simple terrain frame missions without patterns using lidar.

it is making flights by changing the altitude settings of waypoints as below :

[image]

DonLakeFlyer, 2020-04-30:
> I am ready to test on the other hand my english is not very good

No problem.

First thing is you must be using a daily build. 

Also if you are not connected to a vehicle you must tell QGC the firware/vehicle type you are creating the plan for in Mission Start:
[image]

By doing that you should get a different altitude mode option called "Terrain Frame":
[image]

This will allow you to specifiy altitudes are the lidar determined terrain height.

The thing to keep in mind that you are using new QGC features so make sure you do this in a safe enough way that you can bail out easily to take back manual control of the vehicle.

chris89116, 2020-04-30:
ok I test this as soon as the weather is nice around here !

chris89116, 2020-05-01:
On the other hand or where do you find the daily version because currently I am on version 4.0.6 ? Thank you

DonLakeFlyer, 2020-05-01:
https://docs.qgroundcontrol.com/en/releases/daily_builds.html

chris89116, 2020-05-01:
I just tested with the daily version and I already have a remark. 
When the drone is not connected to the PC I have access to the terrain frame for the altitude adjustment on the other hand when I connect the drone this one is not accessible ?

DonLakeFlyer, 2020-05-03:
Seems to be working fine for me. What firmware/vehicle type are you running?

chris89116, 2020-05-03:
I use Arducopter 4.0.3

DonLakeFlyer, 2020-05-03:
Ok, that should be working then. I tested it myself with a daily build and it worked fine. When you are connected to a vehicle can you take a screen shot similar to this from the plan view with the Vehicle info expanded:
[image]

chris89116, 2020-05-03:
When I am connected to my drone :
[image]

DonLakeFlyer, 2020-05-03:
Ok. There is a bug in QGC in that it requires the TERRAIN_ENABLE parameter turned on for it to think terrain frame is supported. So if you turn on TERRAIN_ENABLE it will show up. But I'm not sure how ArduPilot reacts to that if you have lidar. Does it use the terrain heights from lidar or does it use terrain heights from the gcs. I don't know the answer to that.

Anyway, I'll fix the bug and get a new build out.

chris89116, 2020-05-05:
Ok with TERAIN_ENABLE the parameter is show up. i'm going to test like this to see if it will work with the lidar

chris89116, 2020-05-08:
I just tested and I have the same problem caused Mission Planner, my drone does not use this lidar when I make a mission in Auto mode with field monitoring. 
Maybe I should already solve the problem on Mission Planner.

DonLakeFlyer, 2020-05-08:
That would be a firmware problem then not a ground station problem.

chris89116, 2020-05-09:
Yes,I think also

DonLakeFlyer, 2020-11-21:
> Ok. There is a bug in QGC in that it requires the TERRAIN_ENABLE parameter turned on for it to think terrain frame is supported. So if you turn on TERRAIN_ENABLE it will show up. But I'm not sure [...]

This has been changed such that the Terrain Frame selection is always available for ArduPilot vehicles no matter what TERRAIN_ENABLE is set to.

closed 2020-11-21 by DonLakeFlyer