mavlink/qgroundcontrol #8708: Chaining several surveys into one Mission

jstrebel, 2020-04-30:
We fly over agricultural land with a fixed wing using a multispectral camera. With our FW we can fly up to 2 hours. This would enable us to fly several plots one after the other without a stopover. In reality the fields are not directly adjacent. Is it possible to extend QGC in such a way that we chain several so-called survey fields to one mission? What are possible solutions? Thanks Jakob

DonLakeFlyer, 2020-04-30:
Just do it. QGC already supports this. You can have as many survey as you want in a Plan.

closed 2020-04-30 by DonLakeFlyer

Antiheavy, 2020-05-02:
@jstrebel you can use the “blank mission” option to build mission plan from scratch or you can use the “Survey” planning wizard and insert additional surveys into the middle of the mission.