mavlink/qgroundcontrol #8710: Unable to "Select Mission Command" in Planner

NickNothom, 2020-05-01:
## Expected Behavior
I should be able to change the waypoint type. The UI refers to it as "Select Mission Command". 

The [Mission Command Editors](https://docs.qgroundcontrol.com/en/PlanView/PlanView.html) part of the user guide says that picking an item from the categories dropdown should show a menu. It also says that to the right of each command there is a menu that I can use to access additional options such as insert and delete. 

## Current Behavior
When I click one of the categories [Flight Control, Conditionals, Camera] nothing happens. There is no menu on the right. Pressing enter does nothing, and there is no confirm button or anything. 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Make a new mission with some waypoints
2. Attempt to change a waypoint to "Loiter" for example

## System Information
When posting bug reports, include the following information
- Operating System: [Linux Mint 19.3, Windows 10, Android 9]
- QGC Version: [4.0.5]
- QGC build: [stable]
- Flight Controller: [Pixhawk]
- Autopilot (with version): [ArduRover 4.0]

[image]

DonLakeFlyer, 2020-05-01:
Ah crap, that's a nasty bug.

DonLakeFlyer, 2020-05-02:
Wait, I forgot. This was fixed in 4.0.6? You aren't being notified of new releases?

closed 2020-05-02 by DonLakeFlyer

NickNothom, 2020-05-20:
@DonLakeFlyer 
This is still happening in 4.0.6 on all of my machines. 

The caveat that I failed to notice before is that the bug only happens after a vehicle has been connected. I'm using a vehicle with ArduRover 4.0 stable. After I connect my vehicle, the feature remains broken until I restart QGC. Then it works until I connect my vehicle again.