mavlink/qgroundcontrol #8711: Custom application icon in custom build

akontsevich, 2020-05-01:
I see custom build is very flexible however still does not allow to specify custom icon for the application, I see `qgroundcontrol.png` still used everywhere:

- https://github.com/mavlink/qgroundcontrol/blob/master/QGCSetup.pri#L163-L165
- https://github.com/mavlink/qgroundcontrol/blob/master/debian/qgroundcontrol.install#L4
- https://github.com/mavlink/qgroundcontrol/blob/master/custom-example/deploy/qgroundcontrol-start.sh#L8
- https://github.com/mavlink/qgroundcontrol/blob/master/deploy/qgroundcontrol-start.sh#L8
- https://github.com/mavlink/qgroundcontrol/blob/master/deploy/qgroundcontrol.desktop#L6

Same thing for `qgroundcontrol-start.sh` and `qgroundcontrol.desktop`. Is it possible to cusomize their names and content? It is not good to patch main QGC sources outside `/custom` folder. Thanks!

DonLakeFlyer, 2020-05-01:
It is supported for Windows builds, but nothing else yet.

akontsevich, 2020-06-01:
Fixed by  #8790.

closed 2020-06-01 by akontsevich