mavlink/qgroundcontrol #8725: 4.0.6: Ubuntu 20.04 AppImage does not start: libgstgl-1.0.so.0

Fi156, 2020-05-04:
- Ground station operating system:
Thinkpad X201, Kubuntu 20.04

- QGroundControl version and build (daily, stable, self-built from source, etc.)
4.0.6
Version 3.5 is working fine... (but not with Ardupilot 4.x)

When starting the AppImage, following error message will be shown:

/tmp/.mount_GxqLko/QGroundControl: error while loading shared libraries: libgstgl-1.0.so.0: cannot open shared object file: No such file or directory

andrewvoznytsa, 2020-05-04:
@Fi156 can you please check if gstreamer1.0-gl is installed there? If not - please install.

Fi156, 2020-05-04:
sudo apt install gstreamer1.0-gl

will solve the problem, thanks @andrewvoznytsa

andrewvoznytsa, 2020-05-04:
False alarm - closing.

Fi156, 2020-05-10:
For the record:
Its not enough to use the line:
`sudo apt install gstreamer1.0-plugins-bad gstreamer1.0-libav -y`
from the [installation page](https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html)
You have to install gstreamer1.0-gl manually as well...

andrewvoznytsa, 2020-05-10:
Perhaps @hamishwillee can help us with the doc update? gstreamer1.0-gl is mandatory since qmlglsink integration (basically since 2020, +- week/two).

DonLakeFlyer, 2020-05-10:
If this is the case then I don't understand why it isn't packaged in the AppImage?

andrewvoznytsa, 2020-05-10:
We were in this story already - for the Linux we don't ship gstreamer due to unknown historical reasons.

DonLakeFlyer, 2020-05-10:
So let's fix it!

hamishwillee, 2020-05-10:
WHO is going to fix this? I only ask because this keeps coming up, but I think that without an owner to look into it, it will not happen. @DonLakeFlyer nominate someone :-)

I'm testing this now and will (hopefully temporarily) add the dependency in docs.

hamishwillee, 2020-05-12:
Further, I cannot reproduce this - I installed as per the instructions and the app started no problems. Figured you might mean "can't run video" so I also tested using the command:
```
gst-launch-1.0 videotestsrc pattern=ball ! video/x-raw,width=640,height=480 ! x264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5600
```
This was on Ubuntu 20.04 in a VirtualBox VM where I selected to install normal installation with all third party plugins.

All of which changes nothing - who can get the dependencies into the release package?

andrewvoznytsa, 2020-05-12:
Then, most likely, gstreamer1.0-gl was already installed on your system. You can check that with
`apt list | grep gstreamer1.0-gl`

hamishwillee, 2020-05-12:
@andrewvoznytsa Possibly, but a) absolutely clean system, so this would have been part of standard install b) "not the point". Lets get the [Insert swearword of choice] dependencies in the release and never speak of this again :-)

andrewvoznytsa, 2020-05-12:
There are chances that gstreamer1.0-gl was installed due to ubuntu installer decision because of HW config or something like that.

If we add gstreamer1.0-gl to list of dependencies in our manual then we'll actually sync docs with actual code state - code is using this package through calls to qmlglsink. It looks for me as quick fix and has sense in short term.

Regarding shipping gstreamer with QGC - it is not trivial change I think (IMO it will take some time to get it stable on all distros). I can do that but not right now - I think I'll be available for next portion of QGC improvements around September and later.

DonLakeFlyer, 2020-05-12:
> Regarding shipping gstreamer with QGC - it is not trivial change I think

I don't understand this. From what I can tell QGC needs a specific version of gstreamer installed to run correctly . As it currently stands we require end users to install that version if they don't have it on their machine. Somehow this is ok. But including the same gstreamer we ask end users to install manually in the AppImage somehow is a problem? Huh?

andrewvoznytsa, 2020-05-12:
> QGC needs a specific version of gstreamer installed
and compiled for their machine (you can translate that as compiled for their specific Linux distro).

We've to be sure that our compiled gstreamer will work on all distros - it depends on some shared libraries (dlls) and they could be a little bit different on all systems. I'm not sure that it will just work, out of the box - we need to test, at least. Even if we limit distros to Ubuntu - I'm not sure that thing compiled for 18.04 will work on 20.04 and 16.04. If you search bug db you'll quickly notice many 'glibc .so unresolved external' issues due to glibc version changes from 16.04 to 20.04. This glibc is something like msvcrt - basic building block. So it is not as trivial as adding single line to the docs.

DonLakeFlyer, 2020-05-12:
Ok, understood. 

Can we do this:
* The AppImage build we release is said to be supported/tested on Ubuntu 18.04 only. Also since linux seems random as to how users can futz with their systems and screw us. Should/can we bundle the gstreamer we need into the AppImage.
* Can we change the build such that it warns the user if either the gstreamer version we need is not there or if they are running on an unsupported OS/Version combination. This would happen on AppImage runs only. I'm thinking a command line option for it.
* And then for people living on the wild side, they can use qgroundcontrol-start.sh and piece together the packages they need to make QGC work. And "maybe" they can get an out of the box build to work on their unsupported OS/version.
* The final fallback is you just have to build it yourself.

hamishwillee, 2020-05-12:
OK, so for now I added the dependency into docs. BUT there are issues with DNS right now, so might not appear on the built version. Don, did Daniel get back to you on that?

FWIW I think minimal fix would be the second option - notify users of missing dependency(s). I like the first option even more, particularly if we can bundle multiple gstreamer versions to support more releases.

thopiekar, 2020-09-25:
Personally, I believe this is a missing lib in the AppImage. Per logic of AppImages you want to have all dependencies inside the image.

julianoes, 2020-09-28:
I agree with @thopiekar. I just ran into this where video streaming doesn't work for me on Fedora 32:

```
Failed to load plugin '/usr/lib64/gstreamer-1.0/libgstqmlgl.so': /tmp/.mount_ifPYVN/Qt/libs/libQt5Core.so.5: version `Qt_5.13' not found (required by /usr/lib64/gstreamer-1.0/libgstqmlgl.so)
```

julianoes, 2020-09-28:
Related to #7287.

closed 2024-06-02 by HTRamsey