mavlink/qgroundcontrol #8727: Video Problems

DonLakeFlyer, 2020-05-04:
When testing #8726 I noticed those problems and was able to reproduce them in stable as well while testing with SITL ```make px4_sitl_default gazebo_typhoon_h480```:
* With no video source:
  * If you pop the video out to a window you get this ```VideoReceiverLog: Stop called on empty URI ``` spit out at you as the stream is stopped/restarted to move to a window.
* With a video source from typhoon sitl connected
  * Somewhere after connect you get ```GStreamer-CRITICAL **: 09:01:04.343: gst_bus_disable_sync_message_emission: assertion 'bus->priv->num_sync_message_emitters > 0' failed```
  * If you pop out the video to a window, the video never restarts in the window. When you close the window and it goes back to pip to video does restart.
  * Also when you resize the main window with video taking the full qgc window shouldn't it preserve the apsect ratio with bars top/bottom and left/right as needed? So you can see the full video no matter what?

andrewvoznytsa, 2020-05-04:
@MatejFranceskin need your help here. Ping me when you get time for that issue.

farhangnaderi, 2021-02-04:
@Mustafadagkiranlar take a look.

antyagi5, 2021-11-20:
Similar issue as @fxa76 -- same Ubuntu build and PX4/Gazebo simulation

fredowski, 2022-09-23:
During my research to fix video problems on MacOS and Android I found some threading problems and required bugfixes which need to be backported from the gst-plugin 1.18.6. This is in PR https://github.com/mavlink/qgroundcontrol/pull/10420 The problems and fixes are discussed on the PR. Can we put this on the devcall on 29th of September?