mavlink/qgroundcontrol #8743: Reorder Waypoints in Plan

leadingedgetechnologies, 2020-05-12:
Tell us a bit about the feature:

Right now, if I create or import a flight plan, the waypoints are in a fixed order. There is no way to reorder waypoints without deleting and recreating them. It seems that a simple "drag and drop" where I could reorder the "cards" on the right where the list of waypoints is, would be nice. 

Another option would be to "reverse" mission. That is, reverse the order of waypoints. 

I often need to change the flight direction and order within a flight for specific reasons.

Thanks!

jumppp, 2020-11-09:
I also ran into the same problem, don't know if you have solved it. If so, can you give me some ideas？

DonLakeFlyer, 2020-11-19:
Still no support for this

AndKe, 2022-03-30:
Is there a way of moving waypoints to another position ?  - let's say you don't want position 1-2-3 , but to fly 1,3,2  - how can you swap them?

closed 2026-10-05 by github-actions[bot]