mavlink/qgroundcontrol #8744: Need help for a release build

ArmStrong-Z, 2020-05-12:
## Expected Behavior
Hello,I am just begin to develop a qgroundcontrol to add some features for my project,but I have the problem when I build my project in a release mode and finish creating a QGroundControl AppImage, this AppImage which I build myself can't run on other computers even though it has the entire environment for a QGroundControll.AppImage

## Current Behavior
when I run ./QGroundControll.AppImage , it reports as below:
```
(base) ☁  ~  ./QGroundControl.AppImage
Settings location "/home/zzk/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
VideoReceiverLog: unable to find qmlglsink - you need to build it yourself and add to GST_PLUGIN_PATH
MAVLinkLogManagerLog: MAVLink logs directory: "/home/zzk/Documents/QGroundControl/Logs"
System reported locale: QLocale(English, Latin, United States) "en_US"
Loading localization for "en_US"
Qt localization for "en_US" is not present
Map Cache in: "/home/zzk/.cache/QGCMapCache300" / "qgcMapCache.db"
QQmlApplicationEngine failed to load component
qrc:/qml/MainRootWindow.qml:339 Type FlightDisplayView unavailable
qrc:/qml/QGroundControl/FlightDisplay/FlightDisplayView.qml:440 Type FlightDisplayViewVideo unavailable
qrc:/qml/QGroundControl/FlightDisplay/FlightDisplayViewVideo.qml:80 Type QGCVideoBackground unavailable
qrc:/qml/QGroundControl/FlightMap/QGCVideoBackground.qml:19 module "org.freedesktop.gstreamer.GLVideoItem" is not installed

qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
QGCPositionManager error 0

```

## System Information
- Operating System: ubuntu18.04
- QGC Version: v4.0.0.6
- QGC build: self-built from source
- Flight Controller:  Pixhawk

DonLakeFlyer, 2020-05-12:
It seems like you may be building with video disabled. Is that what you wanted?

ArmStrong-Z, 2020-05-13:
> It seems like you may be building with video disabled. Is that what you wanted?
Thanks for your hints!,it's not what I want but how can I solve the problem above

ArmStrong-Z, 2020-05-13:
Thanks for your hints!,it's not what I want but how can I solve the problem above

DonLakeFlyer, 2020-05-13:
Did you follow the dev guide build instructions exactly? It seems like you don't gstreamer installed.

Hrespect, 2020-09-08:
I have the same problem. It can be used normally when not installed the gstreamer before. Anyone has the solution? Thanks!

DonLakeFlyer, 2020-09-14:
> It can be used normally when not installed the gstreamer before.

I'm not quite sure what you mean by this. Are you trying to build your own version of QGC which does not include video streaming?

Darthuhman, 2020-11-02:
Hi there, I build and run daily build of QGC without any problem on Win10 but I can’t build installer file for this.
Note, I install NSIS and in build settings enter ‘CONFIG+=installer’ in qmake!
Thanks a lot for any body to guide me.

DonLakeFlyer, 2020-11-02:
Closing due to no response

closed 2020-11-02 by DonLakeFlyer

jaiswalharsh, 2022-06-03:
Hi!
I am having the same issue and gstreamer is installed but still I get the same error. Any solutions?

Thank you!