mavlink/qgroundcontrol #8746: Joystick button flight modes don't load on startup

almatthew, 2020-05-13:
## Expected Behavior
Joystick flight mode button actions should persist.

## Current Behavior
All flight behavior _except_ flight mode selections persist.

## Steps to Reproduce:
1. Start QGroundControl, connect to an APM autopilot
2. Connect a USB gamepad or joystick
3. Make at least one joystick button action a flight mode change (e.g. Loiter or Guided)
4. Close QGroundControl
5. Open QGroundControl, connect to an APM autopilot
6. Connect a USB gamepad or joystick (if not already connected)
7. In the Joystick settings section, find that the flight mode change actions have been removed

## System Information
- Operating System: OS X & Windows
- QGC Version: Stable_V4.0 (and associated tags)
- QGC build: stable, self-built
- Flight Controller: APM-supported APs
- Autopilot (with version): APM 4.0.3 (though I would imagine PX4 would fail as well)

## Detailed Description
This appears to be due to the way the flight mode button actions are populated at start up in the Joystick class.  Specifically, the flight mode button actions are not populated on startup, even though the flight mode button actions have been saved properly in the ini file.  This appears to be due to the lack of an `activeVehicle`: https://github.com/mavlink/qgroundcontrol/blob/Stable_V4.0/src/Joystick/Joystick.cc#L1070.

## Log Files and Screenshots
Setting the button actions:
[image]

After restarting QGC:
[image]

DonLakeFlyer, 2020-05-13:
Thanks @almatthew.

> Specifically, the flight mode button actions are not populated on startup, even though the flight mode button actions have been saved properly in the ini file.

Ah. You checked the .ini file yourself already and the values are correct in there? That was what I was going to ask you to do first.

DonLakeFlyer, 2020-05-13:
So it's just flight mode change buttons which don't stick then right?

almatthew, 2020-05-13:
Correct, the `.ini` includes the flight modes I selected.  And yes, it appears that its only the flight modes that don't persist, all other actions seem to work fine.

almatthew, 2020-05-13:
@DonLakeFlyer FWIW I put a breakpoint in this method: https://github.com/mavlink/qgroundcontrol/blob/Stable_V4.0/src/Joystick/Joystick.cc#L1050 and found that there are no flight-mode specific actions in `_assignableButtonActions` on first startup.

DonLakeFlyer, 2020-05-13:
Yeah, thanks for digging into this. Definitely screwed up.

EternAlmox, 2020-05-13:
Having the same issue. came from 
https://discuss.px4.io/t/joystick-buttons-are-not-saving-on-android-device/16613/8

closed 2020-08-30 by DonLakeFlyer