mavlink/qgroundcontrol #8754: error: dependent 'qgroundcontrol\libs\mavlink\include\mavlink\v2.0\ardupilotmega\mavlink_msg_efi_status.h' does not exist.

wattsinnovations, 2020-05-14:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Stable release should build with no errors

## Current Behavior
When I build the latest stable release (v4.0), I get the above error

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Clone QGroundControl
2. Checkout the "Stable_V4.0" branch
3. Build

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: 4.0
- QGC build: stable, self built
- Flight Controller: N/A
- Autopilot (with version): N/A

DonLakeFlyer, 2020-05-15:
Do you initialize and update the submodules?

wattsinnovations, 2020-05-15:
> Did you initialize and update the submodules?

Yeah, after cloning I ran `git submodule` and then `git submodule update`. and this appears to have installed some of the submodules

DonLakeFlyer, 2020-05-15:
Delete your entire build directory and build again. You likely have some part of a previous build in there which is getting reused. That header file is truly not part of the version of mavlink the stable build uses. So there is no reason it should be included anywhere because it does not exist.

wattsinnovations, 2020-05-15:
I ended up just recloning the repo, as it turns out that I had accidentally omitted the --recursive flag. However, I am now running into a new issue when I build, where it is saying `C1083: Cannot open include file: './mavlink_msg_efi_status.h': No such file or directory`

DonLakeFlyer, 2020-05-15:
> Delete your entire build directory and build again

Did you do this?

wattsinnovations, 2020-05-15:
Yeah I ran a clean on the entire project before building

DonLakeFlyer, 2020-05-15:
Clean doesn't always do it. Delete the directory.

DonLakeFlyer, 2020-05-15:
"Clean" is associated with your current build and can only clean things it knows about in the current source. If you have an old build in there from some different set of source, sometimes clean will not truly clean everything out. This can happen when switching between branches, rarely. But can be fairly common when a mavlink submodule changes.

DonLakeFlyer, 2020-05-15:
Whenever things really start to make no sense at all. Delete the entire build directory is a good thing to start with otherwise you can end up chasing ghosts.

wattsinnovations, 2020-05-15:
Ah ok, that makes sense. This may be a stupid question as I am new to qt, but where is the build directory located? I'm used to using Electron where the build directory is inside of a subfolder called "dist", but I don't see that nor a folder called "build" or anything that may look like a build folder. Is the build folder the folder that is created next to the original project folder? If so, I deleted that, rebuilt, and got the same error, but now only 2 times instead of the 3 times that I got it initialy.

DonLakeFlyer, 2020-05-15:
Project/Build/General/Build directory:
[image]

DonLakeFlyer, 2020-05-15:
I'm going to close this since I"m pretty sure that is what is going on.

closed 2020-05-15 by DonLakeFlyer

wattsinnovations, 2020-05-16:
Yeah sop that is actually the directory that I have been deleting. I deleted it one more time and rebuilt just to be sure, but yeah im still getting the same `no such file or directory` error as before

DonLakeFlyer, 2020-05-16:
Did you do a git submodule init and then update after you changed to the Stable_V4.0 branch?

wattsinnovations, 2020-05-16:
Yeah, after switching to Stable_V4.0 and deleting the build folder I ran `git submodule init` and then `git submodule update`, and then on build it still threw the same error

wattsinnovations, 2020-05-18:
Any ideas as to what could be causing me to get this error even after running the git commands to update the submodules? I am at a loss at this point as to what I can even try now

DonLakeFlyer, 2020-05-24:
I'm not seeing the same problem with stable on any of my builds. So you'll have to debug through it yourself. The compiler should tell you exactly where is it coming from.