mavlink/qgroundcontrol #8768: Unable to 'Select Mission Command' After Connecting Vehicle

NickNothom, 2020-05-21:
## Expected Behavior
I should be able to change the waypoint type. The UI refers to it as "Select Mission Command". 

**The feature works when I don't have any vehicles connected. As soon as I connect my vehicle, the feature breaks. Restarting QGC fixes the issue.** 

The [Mission Command Editors](https://docs.qgroundcontrol.com/en/PlanView/PlanView.html) part of the user guide says that picking an item from the categories dropdown should show a menu. It also says that to the right of each command there is a menu that I can use to access additional options such as insert and delete. 

## Current Behavior
After connecting a vehicle, when I click one of the categories [Flight Control, Conditionals, Camera] nothing happens. There is no menu on the right. Pressing enter does nothing, and there is no confirm button or anything. 

## Steps to Reproduce:
1. Open QGC with no vehicles connected
2. Make a new mission with some waypoints
3. Change a waypoint to "Loiter" for example. It should work as intended. 
4. Connect Ardurover 4.0 vehicle to QGroundControl
5. Make a new waypoint and try to change its type to "Loiter". At this stage the feature is broken for me, until I restart QGC. 

## System Information
When posting bug reports, include the following information
- Operating System: [Linux Mint 19.3, Windows 10, Android 10]
- QGC Version: [4.0.6]
- QGC build: [stable]
- Flight Controller: [Pixhawk]
- Autopilot (with version): [ArduRover 4.0]

[image]

DonLakeFlyer, 2020-05-25:
I can't reproduce this. Can you reproduce this and then go to Settings/Console and copy the log output and attach here. There may be some clues there.

NickNothom, 2020-05-27:
Here's what shows up:

```
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:263 - "Settings location "C:/Users/nickn/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at C:\projects\qgroundcontrol\src\QGCLoggingCategory.cc:99 - "Filter rules "*Log.debug=false\n""
[E] at :0 - "serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT."
[E] at C:\projects\qgroundcontrol\src\VideoStreaming\VideoReceiver.cc:572 - "Failed because video sink is not set"
[D] at C:\projects\qgroundcontrol\src\Vehicle\MAVLinkLogManager.cc:360 - "MAVLink logs directory: "C:/Users/nickn/Documents/QGroundControl/Logs""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:363 - "System reported locale: QLocale(English, Default, United States) "en_US""
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:437 - "Loading localization for "en_US""
[D] at C:\projects\qgroundcontrol\src\QtLocationPlugin\QGCMapEngine.cpp:164 - "Map Cache in: "C:/Users/nickn/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[E] at qrc:/qml/PreFlightCheckList.qml:37 - "qrc:/qml/PreFlightCheckList.qml:37: TypeError: Cannot call method 'reset' of undefined"
[E] at qrc:/qml/QGroundControl/Controls/ToolStrip.qml:100 - "qrc:/qml/QGroundControl/Controls/ToolStrip.qml:100:21: Unable to assign [undefined] to bool"
[D] at C:\projects\qgroundcontrol\src\QGCApplication.cc:815 - ""v4.0.6""
[D] at C:\projects\qgroundcontrol\src\comm\MAVLinkProtocol.cc:467 - "Temp log "C:/Users/nickn/AppData/Local/Temp/FlightData174094.mavlink""
[E] at :0 - "libpng warning: iCCP: known incorrect sRGB profile"
[D] at qrc:/qml/QGroundControl/Controls/TakeoffItemMapVisual.qml:148 - "mousearea click"
[D] at C:\projects\qgroundcontrol\src\Vehicle\Vehicle.cc:2712 - "_parametersReady true"

```

I did some more digging and it's not happening with a freshly flashed Rover vehicle, it seems to depend on my params. If I flash Copter, then flash Rover again to wipe the params it works fine. After I load my params and reboot, then I get the bug. Attaching my param file. 

[Planner_Bug_Params.zip](https://github.com/mavlink/qgroundcontrol/files/4686122/Planner_Bug_Params.zip)

DonLakeFlyer, 2020-05-27:
It's because the boat vehicle type isn't correctly classified as a Rover. I assume that fact that your params are for a boat is correct.

DonLakeFlyer, 2020-05-27:
Can you verify that this fixes the problem: https://qgroundcontrol.s3-us-west-2.amazonaws.com/builds/Stable_V4.0/QGroundControl-installer.exe

DonLakeFlyer, 2020-05-30:
Daily build is updated as well.

closed 2020-05-30 by DonLakeFlyer

NickNothom, 2020-06-01:
Fix works for me! Thanks