mavlink/qgroundcontrol #8770: [Feature Request] method to easily delete all existing telemetry data and map caching

Antiheavy, 2020-05-23:
In particular, mobile users are not able to easily delete old telemetry logs that build up on their devices.  We'd like a way to easily delete telemetry logs.  It seems the location for this would be in general settings in the Data Persistence section. 

Would a button press work in here?  Or maybe "disable all data persistence" also deletes all previously recorded data?  Or it prompts user and asks if they want to delete or keep previously recorded data?

[image]

DonLakeFlyer, 2020-05-24:
> mobile users are not able to easily delete old telemetry logs that build up on their devices.

That isn't quite correct. Android users have no problem dealing with this since you have access to the file system through the My Files app. Doesn't newest version of iOS have a new "Files" app for this now?

DonLakeFlyer, 2020-05-30:
Map caching delete is already available from Offline Maps.

closed 2026-10-05 by github-actions[bot]