mavlink/qgroundcontrol #8771: Blinking airplanes on a map with ADS-B messages

theDimitron, 2020-05-25:
Hi everyone,

I have an issue with ADS-B massages (or with transponder_report topic). If there are more than one vehicle detected than both images of airplanes on the map star to blink. This all happens like in an antiphase with both vehicles. One of vehicles blinks very fast and is visible only for like a third of a second and then disappears for a second, so it cannot be seen properly. Another vehicle disappears from map for a third of a second and stays on the map a second long. Where does it come from and can it be adjust/controlled? 

- Operating System: Windows 10
- QGC Version: 4.0.6 and I’ve also built from master 
- Flight Controller: Pixhawk 4
- Autopilot: PX4 1.10.2dev

DonLakeFlyer, 2020-05-25:
Are you sure each ADSB vehicle has a unique callsign? It sounds like you may have multiple vehicles reporting the same callsign and then as the new positions come in they bounce around to different positions.

theDimitron, 2020-05-26:
I did some screenshots in Ubuntu.
In MAVlink Inspertor callsigns are different. The message also blinks, and shows first and second aircraft separately with same frequency as on the map. I’m using one fake aircraft with No 00004 and the other is a real FLARM-transmitter. The FLARM-device is attached to Pixhawk, driver is manually written and fake aircraft programmed within it (all INFO messages are for debug). But this blinking issue also comes out when FLARM detects a real aircraft (with no fake aircraft programmed).
I’ve also added Ubuntu-Terminal-output. 

[image]
[image]
[image]
[image]

DonLakeFlyer, 2020-05-30:
Should be fixed in both Stable and Daily builds.

closed 2020-05-30 by DonLakeFlyer